UniversalRobots/Universal_Robots_ROS2_Driver #7: Is ur_robot_driver RobotStateHelper necessary?

AndyZe, 2020-11-13:
As we work toward a minimal demonstration, it would help us to know if the RobotStateHelper is necessary. It looks like there are hardly any references to it in the package.

The reason it's somewhat difficult to port is a simple action client doesn't exist for ROS2 yet.

@fmauch @gavanderhoorn do you have any input?

livanov93, 2020-11-13:
There is an implementation of simple action server here [nav2_main](https://github.com/ros-planning/navigation2/blob/main/nav2_util/include/nav2_util/simple_action_server.hpp) that could maybe be used to preserve the structure of [ros1 robot_state_helper](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/robot_state_helper.cpp).
Another way is to create new action server (without simple encapsulation) and to change the logic within the class.

As mentioned  [here](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/43fd5d0e191ab7fcf679775ae88092ad7795f0c9/ur_robot_driver/README.md#a-note-about-modes) the "set_mode" action server is used to put together calls to multiple services of the [dashboard node](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/src/dashboard_client_ros.cpp)

Other advices are welcome.

fmauch, 2020-11-14:
@AndyZe Yes, this is only a convenience node to provide an action interface to bring the robot into a desired state e.g. for powering it on or recovering after a failure (EM-Stop, protective stop, etc) without using the teach pendant.

closed 2020-11-14 by AndyZe