UniversalRobots/Universal_Robots_ROS2_Driver #18: implement a force/torque state controller

nbbrooks, 2021-01-08:


livanov93, 2021-01-13:
#12  already addressed porting dashboard_client_node to ROS2. All the service servers have to be tested on the real hardware.

TODO:

-  Resource names as parameters
-  Force/torque topic name as parameter
-  Adapt interface to the integrated sensor setup [example](https://github.com/ros-controls/roadmap/blob/master/design_drafts/components_architecture_and_urdf_examples.md#3-industrial-robots-with-integrated-sensor)
- Review if it is Beta milestone beacuse in ROS1 the controller [existed within ros_control](https://github.com/ros-controls/ros_controllers/tree/noetic-devel/force_torque_sensor_controller)l, ros2_control does not support it yet.

livanov93, 2021-01-20:
#12 is addressing this

closed 2021-03-02 by nbbrooks