UniversalRobots/Universal_Robots_ROS2_Driver #21: decide how to publish other stuff towards ROS side,  tool state, digital I/O, robot mode, safety mode, tcp pose (create controllers or to spin another thread to publish this stuff)

nbbrooks, 2021-01-08:


livanov93, 2021-01-13:
Here I am more fan of multiple controller creation which would be based on [controller_interface](https://github.com/ros-controls/ros2_control/blob/master/controller_interface/include/controller_interface/controller_interface.hpp). 
This approach only tends to make a huge list of state and command controllers like:

- 
- TcpPoseStateController
- IOStateController
- IOController
- RobotModeStateController
- SafetyModeStateController
- SpeedSliderController		
- ToolStateController
- ExternalUrScriptController
- ResetMotionController
...
These controllers would interface with driver via one of these [state or command](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/handle.hpp) or [sensor](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/include/hardware_interface/sensor_interface.hpp).

Other option is to hold onto ROS1 approach and spin multiple threads and callbacks within hardware interface to register topics.

AndyZe, 2021-01-14:
^that's my understanding from ros2_control meetings, as well

gavanderhoorn, 2021-01-14:
I would say "officially", drivers should always have used the `*Controller` approach to publish additional information. Even in `ros_control` (ie: the ROS 1 version).

It's just that it was so easy to do it with realtime publishers from the `hardware_interface` itself, no one really bothered.

ps: I really don't like to use the `Controller` suffix for something which doesn't actually control anything. I believe we don't have to any more in `ros2_control`, so it would be great if we could avoid it. It leads to many questions ("ah, so the `joint_state_controller` let's me execute motions?", well, no ..).

livanov93, 2021-02-18:
https://github.com/ros-controls/ros2_control/issues/318

destogl, 2021-02-18:
> ps: I really don't like to use the `Controller` suffix for something which doesn't actually control anything. I believe we don't have to any more in `ros2_control`, so it would be great if we could [...]

Unfortunately, we still have it... I agree fully with you....

@livanov93 
>  This approach only tends to make a huge list of state and command controllers like:

I think we can solve this issue by creating a few general controllers/publishers. Basically, we need only GPIOController and an extension of the `joint_state_controller` which enables the definition of states one wants to have published.

I think this is the most general approach and could make many people happy :)

destogl, 2021-02-23:
Mid-term improvements on the ros2_control side as discussed today in the meeting:

- ros-controls/roadmap#38
- ros-controls/roadmap#39
- ros-controls/ros2_controllers#150

closed 2021-03-02 by nbbrooks