UniversalRobots/Universal_Robots_ROS2_Driver #23: active controller detection - to know which commands are sent to robot (velocity or position)

nbbrooks, 2021-01-08:


destogl, 2021-04-15:
We should wait for ros-controls/ros2_control#348 to get implemented and merged. Otherwise, it this almost impossible to realize in the current architecture.

livanov93, 2021-07-06:
@destogl  Are we good to go with this approach that has already been mentioned:
[prepareSwitch](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/8269e2a50cde4efea25c651c7cd1ded08ee917be/ur_robot_driver/src/hardware_interface.cpp#L557)
[doSwitch](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/8269e2a50cde4efea25c651c7cd1ded08ee917be/ur_robot_driver/src/hardware_interface.cpp#L579)

destogl, 2021-07-09:
> @destogl  Are we good to go with this approach that has already been mentioned: [...]

The method are renamed in ros2_control.

Check [this](https://github.com/ros-controls/ros2_control_demos/blob/master/ros2_control_demo_hardware/src/rrbot_system_multi_interface.cpp) example. It does exactly that.

closed 2021-10-05 by livanov93