UniversalRobots/Universal_Robots_ROS2_Driver #26: dashboard client node

nbbrooks, 2021-01-08:


livanov93, 2021-01-13:
#5 already addressed porting `dashboard_client_node` to ROS2. All the service servers have to be tested on the real hardware.

MGBla, 2021-03-03:
+ /add to log -> works
+ /connect -> [dashboard_client-2] terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'                                                                                                                   
[dashboard_client-2]   what():  parameter 'receive_timeout' has already been declared                                                                                                                                                         
[ERROR] [dashboard_client-2]: process has died [pid 19490, exit code -6,
+ /get_loaded_program -> works
+ /get_robot_mode -> works
+ /get_safety_mode -> works
+ /load_installation -> Not sure yet
+ /load_program -> Not sure yet
+ /popup -> works
+ /program_running -> works
+ /program_saved ->works
+ /program_state ->works
+ /quit -> seems to disconnect the robot but could not verify if all states are correct since connect throws errors
+ /raw_request -> not sure yet what it should do
+ /topic -> has shut down the robot not sure if intended behavior shouldn't quit do this?

MGBla, 2021-03-22:
Pull request ready for review

closed 2021-04-15 by destogl