UniversalRobots/Universal_Robots_ROS2_Driver #28: Run robot calibration

AndyZe, 2021-01-08:


gavanderhoorn, 2021-01-08:
To clarify: the ROS 1 driver *extracts* whatever calibration is present on the controller. There is no actual calibration procedure performed.

Unless you're referring to some other form of calibration in the issue title of course.

destogl, 2021-04-15:
@AndyZe can you give more information on this?

AndyZe, 2021-04-15:
I think this is low priority, but we want to be able to do what the old `ur_calibration` package does.

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/tree/master/ur_calibration

gavanderhoorn, 2021-04-15:
This was one of the main features of the official ROS 1 driver.

Without this, FK based on `JointState` messages can be off by as much as 1 cm.