UniversalRobots/Universal_Robots_ROS2_Driver #65: Use Client Library of the Universal Robots Github

MGBla, 2021-04-08:
Currently we are working on a fork of the Client Library repo with additional ROS2 changes. However, since the Client Library is plain c++ it should support both the ROS1 and ROS2 driver without additional branches.

bblumberg, 2021-04-22:
This is additional information on #93. Unfortunately, I don't have the log file anymore. But, I can provide more details. The official UR client library did build (despite catkin warnings) and I was able to link against it. However, at runtime, using a real ur3e, the controller-manager (?) crashed when I used the sample launch instructions from the documentation. I remember it was a malloc error which is what led me to think that you might have had your own version of the client library as you did. When I set the fake hardware option to true, it worked.Sorry I didn't save the log file.

MGBla, 2021-04-22:
Ok thats weird, I have kind of the exact same setup with an ur3e here. I will try to reproduce the error later on a freshly checked out  workspace

bblumberg, 2021-04-22:
Hey, I rebuilt things using the UR version of the UR client library, and
got the same error as yesterday, but this time i saved the launch.log file.
Hope it helps.

bb

MGBla, 2021-04-22:
As it seems there is currently an issue in the ur_client_library which sound rather similar. There seems to be a problem if the ros-noetic-ur-client-library is installed while simultaneously having the client library in a workspace. Do you have the package installed from a privious use of the ROS1 driver? The current workaround is to remove the apt package but the client library driver team is already working on this.

bblumberg, 2021-04-22:
It seems like a different problem. I am installing in a fresh ros:foxy
container and here is how I am installing  the client library
git clone
https://github.com/UniversalRobots/Universal_Robots_Client_Library.git
urcl-ur && \
cd urcl-ur && \
mkdir build && cd build && \
cmake .. && \
make && \
make install

For what it is worth, when I use the PickNik version of the client library,
I can load controllers, but when I try the joint_trajectory controller, it
sends out commands, but the robot doesn't move. I am getting joint states
etc. so I know that the communication at some level is working.

bb

MGBla, 2021-04-22:
Ok maybe it is the std build process which is failing at some level. Normally i clone the client repo into the colcon workspace and just use ``` colcon build ```  to build the client library and the ros2 driver simultaneously, without installing it into the system.

bblumberg, 2021-04-22:
Thanks, I'll give that a try.

bb

bblumberg, 2021-04-23:
I played around with things a bit. Maybe I was making things more complicated for myself. The client library is part of the driver package, but when I did a colcon build of the driver, I get the error below and I am not enough of a CMake person to easily know how to respond. Any help would be appreciated. Thanks.

bb

CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Findur_client_library.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ur_client_library", but CMake did not find one.

  Could not find a package configuration file provided by "ur_client_library"
  with any of the following names:

    ur_client_libraryConfig.cmake
    ur_client_library-config.cmake

  Add the installation prefix of "ur_client_library" to CMAKE_PREFIX_PATH or
  set "ur_client_library_DIR" to a directory containing one of the above
  files.  If "ur_client_library" provides a separate development package or
  SDK, be sure it has been installed.

MGBla, 2021-04-26:
I'm a little confused about the 'the client library is part of the driver package' part? It should not be inside of the driver package itself but within the colcon src folder. So basically your colcon folder should look somehow similar to this: 
```
├── colcon_ws
│   ├── build
│   ├── install
│   ├── log
│   └── src
│       ├── control_msgs
│       ├── realtime_tools
│       ├── ros2_control
│       ├── ros2_control_demos
│       ├── ros2_controllers
│       ├── Universal_Robots_ROS2_Driver
│       ├── ur_client_library
│       ├── ur_msgs
```

bblumberg, 2021-04-26:
Hi
If I clone the client library into the src directory as you suggest, before
doing the colcon build, I get the following error (which is the same error
if I don't pull in the client library and rely on the vcs to get the
picknik version of the driver):

-- stderr: ur_robot_driver
CMake Error at CMakeLists.txt:19 (find_package):
  By not providing "Findur_client_library.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ur_client_library", but CMake did not find one.

  Could not find a package configuration file provided by
"ur_client_library"
  with any of the following names:
apt update && \
    ur_client_libraryConfig.cmake
    ur_client_library-config.cmake

  Add the installation prefix of "ur_client_library" to CMAKE_PREFIX_PATH or
  set "ur_client_library_DIR" to a directory containing one of the above
  files.  If "ur_client_library" provides a separate development package or
  SDK, be sure it has been installed.

I am also confused because the vcs seems to pull in the picknik version of
the client library (if it doesn't already exist) so none of this should be
necessary.

Sorry to be so much trouble.

bb

MGBla, 2021-04-26:
Could you post the entire log of the build process. Maybe there breaks something while building the client library.  I just tried it on a clean workspace and the following packages inside
```
src
├── control_msgs
├── ros2_control
├── ros2_controllers
├── Universal_Robots_Client_Library
├── Universal_Robots_ROS2_Driver
└── ur_msgs
```
and it build (although with some warnings but no errors).

bblumberg, 2021-04-26:
I have attached a file that has 2 iterations:

1. In the first iteration, I clone into src/Universal_Robots_ROS2_Driver
2. In the second iteration, I clone into src/ur_client_library

MGBla, 2021-05-07:
Fixed in #101

closed 2021-05-07 by MGBla