UniversalRobots/Universal_Robots_ROS2_Driver #69: Scaled joint trajectory controller: `rt_active_goal_` is not declared

AndyZe, 2021-04-09:
On develop branch I get a bunch of build errors related to `rt_active_goal_`. It looks like it is not declared anywhere.

Some of the errors are:

```
scaled_joint_trajectory_controller.cpp:153:11: error: could not convert ‘((ur_controllers::ScaledJointTrajectoryController*)this)->ur_controllers::ScaledJointTrajectoryController::<anonymous>.joint_trajectory_controller::JointTrajectoryController::rt_active_goal_’ from ‘joint_trajectory_controller::JointTrajectoryController::RealtimeGoalHandleBuffer’ {aka ‘realtime_tools::RealtimeBuffer<std::shared_ptr<realtime_tools::RealtimeServerGoalHandle<control_msgs::action::FollowJointTrajectory> > >’} to ‘bool’
  153 |       if (rt_active_goal_)
```

```
scaled_joint_trajectory_controller.cpp:181:27: error: ‘using RealtimeGoalHandleBuffer = class realtime_tools::RealtimeBuffer<std::shared_ptr<realtime_tools::RealtimeServerGoalHandle<control_msgs::action::FollowJointTrajectory> > >’ {aka ‘class realtime_tools::RealtimeBuffer<std::shared_ptr<realtime_tools::RealtimeServerGoalHandle<control_msgs::action::FollowJointTrajectory> > >’} has no member named ‘reset’
  181 |           rt_active_goal_.reset();
```

destogl, 2021-04-09:
Ah yeah, my bad, forgot to rebase this PR. There is a braking change on JTC from yesterday. In 66 it works

AndyZe, 2021-04-13:
This is fixed by now

closed 2021-04-13 by AndyZe