UniversalRobots/Universal_Robots_ROS2_Driver #74: Build instructions and missing .repos.yaml file

bblumberg, 2021-04-13:
I am trying to build the driver per the instructions, but the vcs step fails because .repos.yaml does not exist in the src directory.

 vcs import --skip-existing --input src/Universal_Robots_ROS2_Driver/.repos.yaml src
usage: vcs import [-h] [--input FILE_OR_URL] [--force] [--shallow]
                  [--recursive] [--retry N] [--skip-existing] [--debug] [-w N]
                  [--repos]
                  [path]
vcs import: error: argument --input: can't open 'src/Universal_Robots_ROS2_Driver/.repos.yaml': [Errno 2] No such file or directory: 'src/Universal_Robots_ROS2_Driver/.repos.yaml'

------------------
Thanks,

bb

AndyZe, 2021-04-13:
Hi @bblumberg ,

FYI the package is not quite ready for public use.

destogl, 2021-04-15:
Hi @bblumberg, you should use Universal_*.repos file

AndyZe, 2021-04-15:
It should be ready to test now, @bblumberg!

That line in the README was updated:  `vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos`

This PR (should be merged soon) has the instructions to run hardware. (https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver/pull/85/files)

destogl, 2021-04-15:
Create two *.repo files. One minimal without Moveit and one with "scenarios"

bblumberg, 2021-04-15:
Cool. A couple of things...

1. Just to note, the git clone in the instructions references the develop
branch which no longer exists.

2. I ran into the following problem when compiling the code:
ur_description
--- stderr: warehouse_ros_mongo
CMake Error at CMakeLists.txt:22 (find_package):
  By not providing "Findwarehouse_ros.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "warehouse_ros", but CMake did not find one.

  Could not find a package configuration file provided by "warehouse_ros"
  with any of the following names:

    warehouse_rosConfig.cmake
    warehouse_ros-config.cmake

  Add the installation prefix of "warehouse_ros" to CMAKE_PREFIX_PATH or set
  "warehouse_ros_DIR" to a directory containing one of the above files.  If
  "warehouse_ros" provides a separate development package or SDK, be sure it
  has been installed.
---
Sorry to be so needy :-) But I am excited about trying this out.

bb

destogl, 2021-07-06:
This is closed in #86

closed 2021-07-06 by destogl