UniversalRobots/Universal_Robots_ROS2_Driver #75: Check how to change IO on teach panel when GPIO Controller is running

destogl, 2021-04-13:
Most of the pins seem to work except i cannot set the analog_out[1] current. Another problem i encountered is that after setting an input using the service call i wasn't able to change it in the teach panel. I think as the safety device the teach panel should always be able to override values we set using service calls.

_Originally posted by @MGBla in https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver/issues/66#issuecomment-817984775_

MGBla, 2021-04-13:
Maybe I just used it wront. It is not that obvious which field has to be filled in which way. Can you share insights on how the FUN and PIN values have to be set for the second analog_out?

destogl, 2021-04-15:
Related to #87

MGBla, 2021-05-04:
Fixed in #103

closed 2021-05-17 by MGBla