UniversalRobots/Universal_Robots_ROS2_Driver #77: Speed scaler service uses percent, unlike ROS1

AndyZe, 2021-04-13:
In ROS1, the speed scaling slider used a fraction (0.01-1.0, I believe)

gavanderhoorn, 2021-04-13:
See the [documentation](http://docs.ros.org/en/melodic/api/ur_msgs/html/srv/SetSpeedSliderFraction.html):

```yaml
# Set the speed slider on the teach pendant to the specified value.
#
# Values for 'speed_slider_fraction' must be from [0; 1.0]; values outside this
# valid range will result in an error being returned to the caller.
```

MGBla, 2021-04-19:
Reverted back to range of 0-1.0 and added a warning if range is exceeded. #88

MGBla, 2021-04-19:
@gavanderhoorn I think it would make sense to add an message field to the service response giving the user some info about why the service call failed or are there reasons against it?

gavanderhoorn, 2021-04-19:
I would suggest to discuss this with @fmauch, as he was the one who designed that service definition.

fmauch, 2021-04-20:
Absolutely not against it.

I had in mind, that the only thing going wrong would be the user's input range, but of course another possibility is that the [corresponding library function](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/rtde/rtde_writer.cpp#L65) returns `false`.

I see the benefit of poking the user that the entered value is out-of-range.

destogl, 2021-04-22:
solved in #88

closed 2021-04-22 by destogl