UniversalRobots/Universal_Robots_ROS2_Driver #78: Enable `ur_client_library` dependency

destogl, 2021-04-13:
I commented out the dependency on `ur_client_library` because it can not be found by `rosdep`. We are not exporting it properly. Opening follow-up issue for this.

_Originally posted by @destogl in https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver/issues/71#issuecomment-819032596_

MGBla, 2021-04-28:
Ok i just stumbled over this. Due to this change a clean build of the driver is no longer possible since the ur_robot_driver and the ur_controllers complain that they could not find the package during the build process. Actually that was the main cause of the problem of @bblumberg in #65 . We should quickly revert this since the driver is not build-able with colcon build at the moment, which in my view is the easiest access for beginners.

MGBla, 2021-05-07:
fixed in #97

closed 2021-05-07 by MGBla