UniversalRobots/Universal_Robots_ROS2_Driver #93: documentation...

bblumberg, 2021-04-21:
These may be things you already know, but as someone trying this out, here are somethings that I encountered... Good news: I was able to successfully get things running. I was building within a docker container based on  ros:foxy

1. I had to do a git clone https://... instead of the command in the doc (artifact of being in a docker environment?)
2. I needed to do an apt update before the rosdep
3. I need to build the client library. Initially, I cloned the official client library from UR, but that did not work at runtime (I can send a log file, but I suspect you know this already). I then tried the version of the client library on your site and everything worked.
4. In hardware setup, I think you mean to say that you can either switch to remote mode, or use an external node. As it is you say to switch to remote mode and then describe how to use the external node.
5. I manually had to start the controllers as was indicated in the doc. but then it all worked.
bb

MGBla, 2021-04-22:
@bblumberg  Concerning 3 could you please send the log into #65 . When i tried it last time it worked with the client library of UR. Although it still complains that its missing catkin..

bblumberg, 2021-04-28:
To close the loop:
1. I caused myself trouble by using the official release of the client library because the version that is installed via the vcs needs to be consistent with whatever release you have installed via the instructions associated with the client library. Anyway, it would be good to add explicit instructions regarding the client library.

2. I assumed that I could test things using either the external control or remote mode. It actually only works at the moment using external control. This probably should be noted, since I am probably not alone in assuming I could do either.

3. I actually ran into an issue with URCaps conflicting which meant that my external node didn't work (which I assumed was associated with the driver).

closed 2022-04-25 by fmauch