UniversalRobots/Universal_Robots_ROS2_Driver #94: Add warning to the demo script in readme that it will move in zero configuration

MGBla, 2021-04-27:
We should add a warning for the demo script to specify how the robot will. This way the user is aware of what will happen and can remove cables etc. that might be damaged by the motion

nbbrooks, 2021-04-29:
For each "Example Commands for Testing the Driver" demo
- Add instruction to move arm to home position before running
- Add instruction to check cable routing from base before running
- Add description of expected movement for each demo
- Add gif of demo's movement in rviz

gavanderhoorn, 2021-04-29:
Why not check whether the joints are in a specific configuration (ie: elbow up, rest 0) before starting any motion?

And refuse if that's not the case.

That should not be difficult to check.

MGBla, 2021-04-29:
This makes definitely sense as additional fail safe

closed 2021-08-18 by livanov93