UniversalRobots/Universal_Robots_ROS2_Driver #96: Can not build the driver using the instruction from the readme

urrsk, 2021-04-28:
Hi
I am trying to build the driver by following the instructions from the [Readme.md](https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver/blob/main/README.md)

For testing this out I used the Docker approach and made this Dockerfile

In the first attempt I did exactly as written in the README. --  resulting in that it was missing different set of ros2 packets.
To fix this I added ```apt-get update``` and ```rosdep update```

The next attempt, the docker build command failed on the the colcon build command with a lot of red lines 

The Docker file used:
```Dockerfile
FROM ros:foxy

RUN export COLCON_WS=~/workspace/ros_ws_foxy_ur_driver
RUN mkdir -p $COLCON_WS/src
RUN cd $COLCON_WS
RUN git clone https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver
RUN vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos
# First fix, adding rosdep update
RUN apt-get update && rosdep update 
RUN rosdep install --ignore-src --from-paths src -y -r
RUN /bin/bash -c "source /opt/ros/foxy/setup.bash"
RUN colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
RUN /bin/bash -c "source install/setup.bash"
```

The first lines of the failing output:
```
Step 10/11 : RUN colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
 ---> Running in 2f6ba0ae5221
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 5, in <module>
    import distutils.core
  File "/usr/lib/python3/dist-packages/numpy/distutils/__init__.py", line 6, in <module>
    from . import ccompiler
  File "/usr/lib/python3/dist-packages/numpy/distutils/ccompiler.py", line 13, in <module>
    from distutils.errors import DistutilsExecError, DistutilsModuleError, \
ModuleNotFoundError: No module named 'distutils.errors'
[1.037s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'python_setup_py' in 'lib/python3/dist-packages/numpy': Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 143, in _identify
    retval = extension.identify(_reused_descriptor_instance)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 48, in identify
    config = get_setup_information(setup_py)
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 241, in get_setup_information
    _setup_information_cache[hashable_env] = _get_setup_information(
  File "/usr/lib/python3/dist-packages/colcon_python_setup_py/package_identification/python_setup_py.py", line 281, in _get_setup_information
    result = subprocess.run(
  File "/usr/lib/python3.8/subprocess.py", line 512, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-c', "import sys;from setuptools.extern.packaging.specifiers import SpecifierSet;from distutils.core import run_setup;dist = run_setup(    'setup.py', script_args=('--dry-run',), stop_after='config');skip_keys = ('cmdclass', 'distclass', 'ext_modules', 'metadata');data = {    key: value for key, value in dist.__dict__.items()     if (        not key.startswith('_') and         not callable(value) and         key not in skip_keys and         key not in dist.display_option_names    )};data['metadata'] = {    k: v for k, v in dist.metadata.__dict__.items()     if k not in ('license_files', 'provides_extras')};sys.stdout.buffer.write(repr(data).encode('utf-8'))"]' returned non-zero exit status 1.
```

Okay, maybe the foxy image is missing something to support building with colcon!
So I started over and did the ```Install ROS2 Foxy``` myself, also in a docker. This resulting in the same error.
But I identified that the instructions are missing steps for installing at least: git, python3-vcstool, python3-rosdep and python3-colcon-common-extensions
```Dockerfile
FROM ubuntu:focal

# setup timezone
RUN echo 'Etc/UTC' > /etc/timezone && \
    ln -s /usr/share/zoneinfo/Etc/UTC /etc/localtime && \
    apt-get update && \
    apt-get install -q -y --no-install-recommends tzdata && \
    rm -rf /var/lib/apt/lists/*

# setup environment
ENV LANG C.UTF-8
ENV LC_ALL C.UTF-8

RUN apt-get update && apt-get install -y curl gnupg2 lsb-release && rm -rf /var/lib/apt/lists/*
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg

RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null

# apt install ros-foxy-desktop
RUN apt-get update && apt-get install -y ros-foxy-ros-base
ENV ROS_DISTRO foxy
RUN /bin/bash -c "source /opt/ros/foxy/setup.bash"

#Additional packets needed
RUN apt-get install -y git python3-vcstool python3-rosdep python3-colcon-common-extensions apt-utils

RUN export COLCON_WS=~/workspace/ros_ws_foxy_ur_driver
RUN mkdir -p $COLCON_WS/src
RUN cd $COLCON_WS
RUN git clone https://github.com/PickNikRobotics/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver
RUN vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos
RUN rosdep init && rosdep update
RUN rosdep install --ignore-src --from-paths src -y -r
RUN colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
RUN /bin/bash -c "source install/setup.bash"
```

What do I miss?

gavanderhoorn, 2021-04-28:
I believe images like `ros:foxy` no longer come with `build-essential` et al. installed.

This was done to reduce the sizes of those images.

bblumberg, 2021-04-28:
I ran into the same issue. More generally, providing a working docker file as part of the release is a great idea because many people will probably want this.

urmahp, 2021-05-06:
I think the failing output is related to that not all commands are executed inside the COLCON_WS, changing the docker image to the following below everything will be executed inside the COLCON_WS and the image builds fine. 

Also I think the clone URL in the README should be updated after the repo has been moved from git@github.com:PickNikRobotics/Universal_Robots_ROS2_Driver.git to 
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git

```Dockerfile
FROM ros:foxy

ENV COLCON_WS=~/workspace/ros_ws_foxy_ur_driver
RUN mkdir -p $COLCON_WS/src
RUN cd $COLCON_WS \
	&& git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver \
	&& vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos \
	# First fix, adding rosdep update
	&& apt-get update && rosdep update \
	&& rosdep install --ignore-src --from-paths src -y -r \
	&& /bin/bash -c "source /opt/ros/foxy/setup.bash && colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release" \
	&& echo "source install/setup.bash" >> /root/.bashrc

WORKDIR $COLCON_WS
```

bblumberg, 2021-05-06:
Good catch. I verified that your file works.

urrsk, 2021-05-07:
@urmahp it worked. Thanks
In addition, I added ```rm -rf /var/lib/apt/lists/*``` to minimize the size of the image

urrsk, 2021-05-10:
It works with the docker file from #105.
Therefore I close the issue.

closed 2021-05-10 by urrsk