UniversalRobots/Universal_Robots_ROS2_Driver #107: Log messages on robot connection and disconnection

urmahp, 2021-05-10:
When using the ROS2 driver with external-control URCap I would expect that the ROS2 driver terminal should output log messages once the robot is connected to the driver, currently this is not the case. Currently when the robot connects to the driver no log message is printed, meaning that it is actually not possible to see in the driver terminal that the robot has connected to the driver. If the robot disconnects it also not possible to see that the robot has disconnected in the terminal.

I would expect an output similar to below terminal output, this is the terminal output from the ROS driver, when the robot connects to the ROS driver. 

```bash
[ INFO] [1620633422.400343214]: Robot requested program
[ INFO] [1620633422.400784839]: Sent program to robot
[ INFO] [1620633422.418806644]: Robot connected to reverse interface. Ready to receive control commands.
```

I think this is related to that only error and warning messages are currently printed from the client library in the ROS2 driver. Because I can see error and warning messages from the client library in the ROS2 drivers terminal output, but not any log messages above warning level.

MGBla, 2021-05-21:
Ok this is rather weird. According to the [launch files](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/9106b97ea61942f48472e6395489b5d10182422c/ur_bringup/launch/ur_control.launch.py#L232-L233)  the input of the client library should be forwarded to the screen. When adding std::out to the client library code the get printed as well but somehow it seems the UR logging is not captured correctly

urmahp, 2021-05-25:
I think it is related to that currently the client library uses [console bridge](https://github.com/ros/console_bridge) when logging messages, if console bridge is found in the system when building. See [description in the client library](https://github.com/UniversalRobots/Universal_Robots_Client_Library#console_bridge). As default log level is set to warning in the console bridge.

A quick fix could be to change the log level to INFO, when using console bridge.
```c++
console_bridge::setLogLevel(console_bridge::CONSOLE_BRIDGE_LOG_INFO);
```

I think that the best solution is to use the [loghandler interface](https://github.com/UniversalRobots/Universal_Robots_Client_Library#console_bridge) to create a log handler to translate the log messages from the client library into ROS2 log messages. I will be happy to look into this, as I will most likely be implementing something similar in the ROS driver, when I have the time.

destogl, 2021-07-06:
Closed in #126

closed 2021-07-06 by destogl