UniversalRobots/Universal_Robots_ROS2_Driver #113: Pipeline Producer overflowed...

bblumberg, 2021-05-17:
As an experiment, I tried running the beta UR ROS2 driver on an Nvidia Xavier AGX (ARM processor), both in a docker container as well as standalone. In both cases, I get a runtime warning that "no realtime capabilities found", and then once the external node on the robot starts up, I get an alternating " Pipeline producer overflowed!" from line 412 of Universal_Robots_Client_Library/include/ur_client_library/comm/pipeline.h followed by an attempt to reconnect, and then the overflow and so on. Note, I have also seen the overflow error on my Linux machine, but it is much rarer.

I originally posted it as a  URCLientLibrary issue (#59), and got this response from Felix, so I am filing this issue:

> This is indeed a problem with the ROS2 driver.
> 
> The client library stores packages received from the robot inside a queue which has to be emptied (consumed) by a second party. If the queue fills up without being emptied, the mentioned warning is printed.
> 
> It is the driver's job to empty the queue, which should happen here. I'll close this, as it looks like a ROS2 driver issue. Please open an issue inside the ROS2 driver repo linking to this issue. If any news concerning the library come up hunting this down, feel free to comment or reopen this one.

MGBla, 2021-05-21:
I encountered this problem recently while fixing the GPIOs. Here we wait for the response of the client library in a different thread so we can return the correct success flag to the service call. In the old version occurred some errors which lead to an endless waiting which in turn prevented the pipeline from being cleared. Could you try it again with the current version and verify that the error still occurs @bblumberg

bblumberg, 2021-05-25:
I successfully ran the ROS2 driver in a docker container on my nvidia xavier agx.  It seems like the issue is still present, though.  For example, if I bring up the driver in a terminal, ctrl z and then run it in the background, I get the unrecoverable overflow error.  By contrast, when I brought it up in one terminal, and then switched to another terminal to run the demo it all worked. Clearly, the latter way of bringing up the demo is the preferred way, but it seems like the system should be robust enough to deal with the interruption caused by the first way of running the system. I am also using a realtime kernel now which might be helping as well.

destogl, 2022-02-01:
Please check the latest main now when  #281 is merged.

fmauch, 2022-05-06:
Closing this since there have been changes tackling this and it hasn't shown up recently.

closed 2022-05-06 by fmauch