UniversalRobots/Universal_Robots_ROS2_Driver #114: Doesn't build

bblumberg, 2021-05-20:
As of today, scaled_joint_trajectory_controller doesn't build. I tracked it down to changes made to  joint_trajectory_controller in
https://github.com/ros-controls/ros2_controllers.git

halt() disappeared and a few instance variables have been renamed.

gavanderhoorn, 2021-05-21:
It would be good to pin versions of dependencies in [Universal_Robots_ROS2_Driver.repos](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/9106b97ea61942f48472e6395489b5d10182422c/Universal_Robots_ROS2_Driver.repos) to prevent this from happening in the future.

gexter192, 2021-05-21:
Hotfix: 

In `Universal_Robots_ROS2_Driver.repos` change version of  `ros2_controllers` to last working version:

```
  ros2_controllers:
    type: git
    url: https://github.com/ros-controls/ros2_controllers.git
    version: master
```

```
  ros2_controllers:
    type: git
    url: https://github.com/ros-controls/ros2_controllers.git
    version: 0.2.1
```

Should I make a pull request?

gavanderhoorn, 2021-05-21:
@gexter192: the worst that can happen is that it won't get merged based on technical merits of your proposal.

So yes: please submit a PR with your proposed solution.

I'd even suggest to pin other dependencies as well, but I'd also suggest to do that in a separate commit, so it can easily be removed if that would be undesirable.

PR #115 (2021-05-21): change version of ros2_controllers in Universal_Robots_ROS2_Driver.repos to 0.2.1 - project builds now
fixes #114 

change version of ros2_controllers in Universal_Robots_ROS2_Driver.repos to 0.2.1 - project builds now
closed 2021-05-21 by MGBla