UniversalRobots/Universal_Robots_ROS2_Driver #116: Driver not able to connect to real robot

mat-l, 2021-05-21:
I have a fresh installation of ROS Foxy on Ubuntu 20.04 with the latest URcap for a real UR5e.

Currently I am not able to connect ROS with the real robot, please find the terminal log attached. any ideas how to fix it?
 
```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.2.100 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/z0018773/.ros/log/2021-05-21-14-04-38-048374-sbrc87201-162798
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [162802]
[INFO] [robot_state_publisher-2]: process started with pid [162804]
[INFO] [rviz2-3]: process started with pid [162806]
[INFO] [spawner.py-4]: process started with pid [162808]
[INFO] [spawner.py-5]: process started with pid [162810]
[INFO] [spawner.py-6]: process started with pid [162812]
[INFO] [spawner.py-7]: process started with pid [162814]
[INFO] [spawner.py-8]: process started with pid [162816]
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link base had 0 children
[robot_state_publisher-2] Link base_link_inertia had 1 children
[robot_state_publisher-2] Link shoulder_link had 1 children
[robot_state_publisher-2] Link upper_arm_link had 1 children
[robot_state_publisher-2] Link forearm_link had 1 children
[robot_state_publisher-2] Link wrist_1_link had 1 children
[robot_state_publisher-2] Link wrist_2_link had 1 children
[robot_state_publisher-2] Link wrist_3_link had 1 children
[robot_state_publisher-2] Link flange had 1 children
[robot_state_publisher-2] Link tool0 had 0 children
[robot_state_publisher-2] [INFO] [1621598678.397783993] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1621598678.397877781] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1621598678.397887122] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1621598678.397891532] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1621598678.397895274] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1621598678.397899009] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1621598678.397902490] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1621598678.397905939] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1621598678.397909440] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1621598678.397912850] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1621598678.397916389] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1621598678.416411334] [URPositionHardwareInterface]: Starting ...please wait...
[spawner.py-8] [INFO] [1621598678.560014199] [spawner_joint_trajectory_controller]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1621598678.579233135] [spawner_io_and_status_controller]: Waiting for /controller_manager services
[spawner.py-6] [INFO] [1621598678.585511638] [spawner_speed_scaling_state_broadcaster]: Waiting for /controller_manager services
[rviz2-3] [INFO] [1621598678.789000078] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1621598678.789120459] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-3] [INFO] [1621598678.838288232] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[... 321 lines cut ...]
[ros2_control_node-1]          at line 45 in /home/z0018773/workspace/ros_ws_foxy_ur_driver/src/Universal_Robots_Client_Library/src/ur/calibration_checker.cpp
[rviz2-3] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "shoulder_link" passed to canTransform argu
```

mat-l, 2021-05-21:
If I connect the ROS to a UR5e in a Virtual Box running on another machine, the robots position gets read once, but the the data stream is interrupted and I cannot send a new pose to the robot as well.

You can find the logs following:

```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.2.107 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/z0018773/.ros/log/2021-05-21-14-10-08-960356-sbrc87201-163927
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [163931]
[INFO] [robot_state_publisher-2]: process started with pid [163933]
[INFO] [rviz2-3]: process started with pid [163935]
[INFO] [spawner.py-4]: process started with pid [163937]
[INFO] [spawner.py-5]: process started with pid [163939]
[INFO] [spawner.py-6]: process started with pid [163941]
[INFO] [spawner.py-7]: process started with pid [163943]
[INFO] [spawner.py-8]: process started with pid [163945]
[robot_state_publisher-2] Parsing robot urdf xml string.
[robot_state_publisher-2] Link base had 0 children
[robot_state_publisher-2] Link base_link_inertia had 1 children
[robot_state_publisher-2] Link shoulder_link had 1 children
[robot_state_publisher-2] Link upper_arm_link had 1 children
[robot_state_publisher-2] Link forearm_link had 1 children
[robot_state_publisher-2] Link wrist_1_link had 1 children
[robot_state_publisher-2] Link wrist_2_link had 1 children
[robot_state_publisher-2] Link wrist_3_link had 1 children
[robot_state_publisher-2] Link flange had 1 children
[robot_state_publisher-2] Link tool0 had 0 children
[robot_state_publisher-2] [INFO] [1621599009.318596339] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1621599009.318674798] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1621599009.318683313] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1621599009.318688232] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1621599009.318692364] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1621599009.318696491] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1621599009.318700704] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1621599009.318704795] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1621599009.318708829] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1621599009.318712859] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1621599009.318716762] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1621599009.326041174] [URPositionHardwareInterface]: Starting ...please wait...
[spawner.py-4] [INFO] [1621599009.468037558] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-6] [INFO] [1621599009.474496068] [spawner_speed_scaling_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-5] [INFO] [1621599009.476014094] [spawner_io_and_status_controller]: Waiting for /controller_manager services
[rviz2-3] [INFO] [1621599009.669394740] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1621599009.669517237] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-3] [INFO] [1621599009.721199866] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] Parsing robot urdf xml string.
[rviz2-3] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[rviz2-3] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.10/src/buffer_core.cpp
[... cut ...]
[INFO] [robot_state_publisher-2]: process has finished cleanly [pid 163933]
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
[ERROR] [ros2_control_node-1]: process has died [pid 163931, exit code -2, cmd '/home/z0018773/workspace/ros_ws_foxy_ur_driver/install/controller_manager/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_jseqfsp2 --params-file /home/z0018773/workspace/ros_ws_foxy_ur_driver/install/ur_bringup/share/ur_bringup/config/ur_controllers.yaml'].
```

urrsk, 2021-05-27:
Can you verify you are able to ping the robot from the driver side as well as ping from the driver to the robot side?

mat-l, 2021-05-31:
I was able to ping the robot.

Besides I am testing ROS2 Foxy with your driver in a VM set up to get to know the basics. Here it was able to read the robots pose for once, but then it also throws the error `[ros2_control_node-1] Warning: Failed to read from stream, reconnecting in 64 seconds...`

Additionally, as you can see in the screenshot, the "plan & execude" does not work. The robot in the VM is not moving. 
[image]

I also published a message at the topic "/ur_hardware_interface/script_command" from the command line, but again the robot is not moving: `ros2 topic pub /ur_hardware_interface/script_command std_msgs/String "data: 'movel([0.5,0.5,0.5,3.14,-1.57,1.57])'"`

urrsk, 2021-06-01:
Can you try to change the log level for the client_library, try again and return the terminal output as well as the simulated robot's log?
See how to change the log level for the client_library here: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/107#issuecomment-847636324

mat-l, 2021-06-01:
We pulled again the latest sources from your repo and now the connection to the real robot is working and we can move the robot. We didn't change anything else, so maybe there was some commit that fixed it already. :)

closed 2021-06-01 by mat-l

NebuDa, 2021-08-20:
Hello @mat-l , I have the same issue with the Simulator and at this time no Access to a real robot. I can change the position in the simulator and RViz shows the movement, but I cannot send a trajectory from ROS to the URSim. Did you fix the problem with the simulator, too? I have the latest URCap and driver sources. 
Running Foxy on Ubuntu like you.
To know, that the driver does not really work with the simulator will help, also.