UniversalRobots/Universal_Robots_ROS2_Driver #117: Dashboard client services unavailable when using ur_control.launch.py

urmahp, 2021-05-21:
When using the launch file [ur_control.launch.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_bringup/launch/ur_control.launch.py) nothing happens when calling the services from DashboardClientROS. Even though the services are available. 

I am launching the driver with the following command. 
```bash
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=127.0.0.1
```
When calling the service to e.g. power on the robot, it waits for the service to come available. 
```bash
ros2 service call /ur_dashboard_client/power_on std_srvs/srv/Trigger
waiting for service to become available...
requester: making request: std_srvs.srv.Trigger_Request()
```
And it just sits there until I terminate it. 

When listing the services available, the service is available. 
```bash
root@ur-li-95qgtn2:/~/workspace/ros_ws_foxy_ur_driver# ros2 service list
/controller_manager/configure_and_start_controller
/controller_manager/configure_controller
/controller_manager/describe_parameters
/controller_manager/get_parameter_types
/controller_manager/get_parameters
/controller_manager/list_controller_types
/controller_manager/list_controllers
/controller_manager/list_hardware_interfaces
/controller_manager/list_parameters
/controller_manager/load_and_configure_controller
/controller_manager/load_and_start_controller
/controller_manager/load_controller
/controller_manager/reload_controller_libraries
/controller_manager/set_parameters
/controller_manager/set_parameters_atomically
/controller_manager/switch_controller
/controller_manager/unload_controller
/force_torque_sensor_broadcaster/describe_parameters
/force_torque_sensor_broadcaster/get_parameter_types
/force_torque_sensor_broadcaster/get_parameters
/force_torque_sensor_broadcaster/list_parameters
/force_torque_sensor_broadcaster/set_parameters
/force_torque_sensor_broadcaster/set_parameters_atomically
/io_and_status_controller/describe_parameters
/io_and_status_controller/get_parameter_types
/io_and_status_controller/get_parameters
/io_and_status_controller/list_parameters
/io_and_status_controller/set_io
/io_and_status_controller/set_parameters
/io_and_status_controller/set_parameters_atomically
/io_and_status_controller/set_speed_slider
/joint_state_broadcaster/describe_parameters
/joint_state_broadcaster/get_parameter_types
/joint_state_broadcaster/get_parameters
/joint_state_broadcaster/list_parameters
/joint_state_broadcaster/set_parameters
/joint_state_broadcaster/set_parameters_atomically
/joint_trajectory_controller/describe_parameters
/joint_trajectory_controller/get_parameter_types
/joint_trajectory_controller/get_parameters
/joint_trajectory_controller/list_parameters
/joint_trajectory_controller/set_parameters
/joint_trajectory_controller/set_parameters_atomically
/robot_state_publisher/describe_parameters
/robot_state_publisher/get_parameter_types
/robot_state_publisher/get_parameters
/robot_state_publisher/list_parameters
/robot_state_publisher/set_parameters
/robot_state_publisher/set_parameters_atomically
/speed_scaling_state_broadcaster/describe_parameters
/speed_scaling_state_broadcaster/get_parameter_types
/speed_scaling_state_broadcaster/get_parameters
/speed_scaling_state_broadcaster/list_parameters
/speed_scaling_state_broadcaster/set_parameters
/speed_scaling_state_broadcaster/set_parameters_atomically
/ur_dashboard_client/add_to_log
/ur_dashboard_client/brake_release
/ur_dashboard_client/clear_operational_mode
/ur_dashboard_client/close_popup
/ur_dashboard_client/close_safety_popup
/ur_dashboard_client/connect
/ur_dashboard_client/describe_parameters
/ur_dashboard_client/get_loaded_program
/ur_dashboard_client/get_parameter_types
/ur_dashboard_client/get_parameters
/ur_dashboard_client/get_robot_mode
/ur_dashboard_client/get_safety_mode
/ur_dashboard_client/list_parameters
/ur_dashboard_client/load_installation
/ur_dashboard_client/load_program
/ur_dashboard_client/pause
/ur_dashboard_client/play
/ur_dashboard_client/popup
/ur_dashboard_client/power_off
/ur_dashboard_client/power_on
/ur_dashboard_client/program_running
/ur_dashboard_client/program_saved
/ur_dashboard_client/program_state
/ur_dashboard_client/quit
/ur_dashboard_client/raw_request
/ur_dashboard_client/restart_safety
/ur_dashboard_client/set_parameters
/ur_dashboard_client/set_parameters_atomically
/ur_dashboard_client/shutdown
/ur_dashboard_client/stop
/ur_dashboard_client/unlock_protective_stop
```

If I instead use the launch file [ur_dashboard_client.launch.py](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_bringup/launch/ur_dashboard_client.launch.py) the services are working as expected I can e.g. power on the robot with a service call.

MGBla, 2021-05-21:
Ok at least I found the error. In the current version we create a new node for the dashboard inside the hardware interface and hand this over to the dashboard client similar to the dashboard_client_node which is used in the ur_dashboard_client.launch.py. However in this case we do not spin the node which in turn leads to not executed callbacks. The problem here is that a spin is blocking the rest of the hardware interface. I think the easiest solution would remove the dashboard_client from the hardware interface and instead just start it in the launch file.

urmahp, 2021-05-25:
> Ok at least I found the error. In the current version we create a new node for the dashboard inside the hardware interface and hand this over to the dashboard client similar to the [...]

Seems like a good solution.

PR #119 (2021-05-25): Removed dashboard client from hardware interface
Fixes #117
closed 2021-05-25 by MGBla