UniversalRobots/Universal_Robots_ROS2_Driver #120: scaled_joint_trajectory_controller missing implementation

mdragusu, 2021-05-25:
Hello, I tried to build the driver according to the readme
I used a container
```
FROM ubuntu:20.04

ENV DEBIAN_FRONTEND=noninteractive 
# update packages and install dependencies
RUN apt-get update && apt-get install -y \
    wget

RUN apt-get update && apt-get install -q -y \
    bash-completion \
    dirmngr \
    gnupg2 \
    lsb-release \
    python3-pip \
    curl 

# Setup environment
ENV LANG C.UTF-8
ENV LC_ALL C.UTF-8

RUN pip3 install colcon-common-extensions rosdep vcstool 

RUN apt-get update && apt-get install --no-install-recommends -y \
    autoconf \
    automake \
    cmake \
    libtool \
    libltdl-dev \
    git \
    mercurial \
    && rm -rf /var/lib/apt/lists/*
RUN apt update && apt install curl gnupg2 lsb-release
RUN curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key  -o /usr/share/keyrings/ros-archive-keyring.gpg
RUN echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" | tee /etc/apt/sources.list.d/ros2.list > /dev/null
RUN apt update && apt install -q -y ros-foxy-ros-base

```

and the following steps

```
docker run -t -v "$PWD:/work" ros2_ur5 .
cd /work/
mkdir -p /work/src/
git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver
vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos
source /opt/ros/foxy/setup.sh
rosdep init
rosdep update
rosdep install --ignore-src --from-paths src -y -r
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

```
The problem I hit is:
```
--- stderr: ur_controllers                               
/work/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update()’:
/work/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:58:10: error: ‘is_halted’ was not declared in this scope; did you mean ‘is_halted_’?
   58 |     if (!is_halted) {
      |          ^~~~~~~~~
      |          is_halted_
/work/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:59:7: error: ‘halt’ was not declared in this scope
   59 |       halt();
      |       ^~~~
/work/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:93:38: error: ‘joint_position_state_interface_’ was not declared in this scope; did you mean ‘joint_state_interface_’?
   93 |     state_current.positions[index] = joint_position_state_interface_[index].get().get_value();
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                      joint_state_interface_
/work/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:94:39: error: ‘joint_velocity_state_interface_’ was not declared in this scope; did you mean ‘has_velocity_state_interface_’?
   94 |     state_current.velocities[index] = joint_velocity_state_interface_[index].get().get_value();
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                       has_velocity_state_interface_
/work/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:128:9: error: ‘joint_position_command_interface_’ was not declared in this scope; did you mean ‘has_position_command_interface_’?
  128 |         joint_position_command_interface_[index].get().set_value(state_desired.positions[index]);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         has_position_command_interface_
make[2]: *** [CMakeFiles/ur_controllers.dir/build.make:63: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/ur_controllers.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ur_controllers [2.44s, exited with code 2]

```
Is this a known problem? Should I use a different working branch?

bblumberg, 2021-05-25:
Is it possible that you are using a cached version of the repository? Check that you have this in your Universal_Robots....repos file:

ros2_controllers:    
  type: git   
  url: https://github.com/ros-controls/ros2_controllers.git
  version: 0.2.1

bblumberg, 2021-05-25:
Ok, but it seems to be pulling in a different version (which is why you are seeing the error that you are). The other thing to try is to use $COLCON_WS as is used in the instructions. For whatever reason this seems important to set that environment variable (as in when I did things manually without setting $COLCON_WS, I couldn't get things to build in docker, even though I was sure I was doing the equivalent operations). Sorry I can't be more helpful.

mdragusu, 2021-05-25:
it seem a pull was all I needed
```
From https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver
   c5b84db..38a467e  main       -> origin/main
Updating c5b84db..38a467e
Fast-forward
 ur_bringup/launch/ur_control.launch.py                         | 10 ++++++++++
 ur_robot_driver/doc/features.md                                | 53 +++++++++++++++++++++++------------------------------
 ur_robot_driver/include/ur_robot_driver/hardware_interface.hpp |  1 -
 ur_robot_driver/src/hardware_interface.cpp                     |  4 ----
 4 files changed, 33 insertions(+), 35 deletions(-)

```
Now it's ok
Finished <<< ros2_controllers [1.03s]                                                                         
Finished <<< ur_controllers [19.8s]                              

Summary: 30 packages finished [3min 46s]

Thank you

closed 2021-05-25 by mdragusu