UniversalRobots/Universal_Robots_ROS2_Driver #127: Colcon build command error in installation

ppascotto, 2021-06-09:
I am trying to install the driver and I am running into CMake error on the colcon build step of the installation instructions. I am using Ubuntu focal 20.04 and foxy. 

`<pre>$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting &gt;&gt;&gt; control_msgs
Starting &gt;&gt;&gt; controller_manager_msgs
Starting &gt;&gt;&gt; ros2_control_test_assets
Starting &gt;&gt;&gt; ros2_control_test_nodes
Starting &gt;&gt;&gt; ur_client_library
Starting &gt;&gt;&gt; ur_dashboard_msgs
Starting &gt;&gt;&gt; ur_msgs
Starting &gt;&gt;&gt; ur_description
Starting &gt;&gt;&gt; ur_moveit_config
Finished &lt;&lt;&lt; ros2_control_test_assets [0.40s]                                                            
Finished &lt;&lt;&lt; ros2_control_test_nodes [0.46s]
Finished &lt;&lt;&lt; ur_description [0.50s]                                                                      
Finished &lt;&lt;&lt; ur_moveit_config [0.56s]                                                                    
Finished &lt;&lt;&lt; ur_client_library [0.64s]
Finished &lt;&lt;&lt; controller_manager_msgs [1.43s]                                                               
Finished &lt;&lt;&lt; ur_dashboard_msgs [1.63s]                                                                    
Finished &lt;&lt;&lt; ur_msgs [1.74s]                                                                
Finished &lt;&lt;&lt; control_msgs [1.99s]                   
Starting &gt;&gt;&gt; hardware_interface
Finished &lt;&lt;&lt; hardware_interface [1.02s]                      
Starting &gt;&gt;&gt; controller_interface
Starting &gt;&gt;&gt; ros2_control_demo_hardware
Starting &gt;&gt;&gt; transmission_interface
Finished &lt;&lt;&lt; ros2_control_demo_hardware [1.04s]                            
Starting &gt;&gt;&gt; ros2_control_demo_robot
Finished &lt;&lt;&lt; transmission_interface [1.05s]
Finished &lt;&lt;&lt; ros2_control_demo_robot [0.15s]                                   
Starting &gt;&gt;&gt; ros2_control_demos                              
Finished &lt;&lt;&lt; controller_interface [1.26s]
Starting &gt;&gt;&gt; controller_manager
Finished &lt;&lt;&lt; ros2_control_demos [0.15s]                                                                  
Finished &lt;&lt;&lt; controller_manager [1.40s]                      
Starting &gt;&gt;&gt; forward_command_controller
Starting &gt;&gt;&gt; joint_state_broadcaster
Starting &gt;&gt;&gt; joint_trajectory_controller
Starting &gt;&gt;&gt; diff_drive_controller
Starting &gt;&gt;&gt; ros2controlcli
Starting &gt;&gt;&gt; ur_bringup                                                    
Starting &gt;&gt;&gt; ur_robot_driver
--- stderr: joint_trajectory_controller                                          
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing &quot;Findangles.cmake&quot; in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by &quot;angles&quot;, but
  CMake did not find one.

  Could not find a package configuration file provided by &quot;angles&quot; with any
  of the following names:

    anglesConfig.cmake
    angles-config.cmake

  Add the installation prefix of &quot;angles&quot; to CMAKE_PREFIX_PATH or set
  &quot;angles_DIR&quot; to a directory containing one of the above files.  If &quot;angles&quot;
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   &lt;&lt;&lt; joint_trajectory_controller [0.28s, exited with code 1]
Aborted  &lt;&lt;&lt; ur_bringup [0.25s]
Aborted  &lt;&lt;&lt; ros2controlcli [0.45s]                                                         
Aborted  &lt;&lt;&lt; forward_command_controller [0.99s]                                             
Aborted  &lt;&lt;&lt; diff_drive_controller [1.01s]
Aborted  &lt;&lt;&lt; ur_robot_driver [9.39s]                                                          
Aborted  &lt;&lt;&lt; joint_state_broadcaster [15.3s]                                    

Summary: 16 packages finished [21.3s]
  1 package failed: joint_trajectory_controller
  6 packages aborted: diff_drive_controller forward_command_controller joint_state_broadcaster ros2controlcli ur_bringup ur_robot_driver
  3 packages had stderr output: diff_drive_controller forward_command_controller joint_trajectory_controller
  7 packages not processed
</pre>`

bblumberg, 2021-06-10:
Did you build foxy from src or are you using the debian? I ran into the same problem when I built from src, but no problem from the debian.

ppascotto, 2021-06-10:
I rebuilt foxy from debian and the packages were able to be finished. I will close this issue, thank you.

closed 2021-06-10 by ppascotto