UniversalRobots/Universal_Robots_ROS2_Driver #128: Basic teleop and control

mdragusu, 2021-06-11:
I have been playing around with the examples provided, but there is little else to do.
Is there a recommended premade teleop application that works with your setup?
If not do you plan on providing a teleop example or a go-to example (using inverse kinematics)?

destogl, 2021-06-15:
@mdragusu how would you like to control the robot in teleop scenario? There is a possibility to run MoveIt Servo with this driver.

mdragusu, 2021-06-16:
I would like to control it either with a keyboard or xbox360/ps4 controller from a remote computer without a GUI.
Or of possible to ask it to go to x,y,z and let the inverse kinematics do the work, again with not GUI.
Do you have examples like that?

AndyZe, 2021-06-16:
It sounds like you should investigate the[ MoveIt2 tutorials](http://moveit2_tutorials.picknik.ai/). The driver isn't what handles this aspect, even though we left a few examples with the package.

Here's a good tutorial for the XBox controller:  http://moveit2_tutorials.picknik.ai/doc/realtime_servo/realtime_servo_tutorial.html

And here's a good tutorial on IK. There are several options for doing this:  http://moveit2_tutorials.picknik.ai/doc/move_group_interface/move_group_interface_tutorial.html

closed 2021-06-16 by AndyZe