UniversalRobots/Universal_Robots_ROS2_Driver #131: Build failing with ros2_controllers installed from apt

sanatmharolkar, 2021-06-20:
Hi, 

I already had the ros2_control packages installed from apt using the command "sudo apt install ros-foxy-ros2-control ros-foxy-ros2-controllers ros-foxy-control-msgs". After cloning the UR driver packages, I made sure the dependencies were satisfied using rosdep. So then I tried to build the UR driver packages in my colcon workspace. However the build fails with the following message:

```
/home/sanat/colcon_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp: In member function ‘virtual controller_interface::return_type ur_controllers::ScaledJointTrajectoryController::update()’:
/home/sanat/colcon_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:58:10: error: ‘is_halted’ was not declared in this scope; did you mean ‘is_halted_’?
   58 |     if (!is_halted) {
      |          ^~~~~~~~~
      |          is_halted_
/home/sanat/colcon_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:59:7: error: ‘halt’ was not declared in this scope
   59 |       halt();
      |       ^~~~
/home/sanat/colcon_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:93:38: error: ‘joint_position_state_interface_’ was not declared in this scope; did you mean ‘joint_state_interface_’?
   93 |     state_current.positions[index] = joint_position_state_interface_[index].get().get_value();
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                      joint_state_interface_
/home/sanat/colcon_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:94:39: error: ‘joint_velocity_state_interface_’ was not declared in this scope; did you mean ‘has_velocity_state_interface_’?
   94 |     state_current.velocities[index] = joint_velocity_state_interface_[index].get().get_value();
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                       has_velocity_state_interface_
/home/sanat/colcon_ws/src/Universal_Robots_ROS2_Driver/ur_controllers/src/scaled_joint_trajectory_controller.cpp:128:9: error: ‘joint_position_command_interface_’ was not declared in this scope; did you mean ‘has_position_command_interface_’?
  128 |         joint_position_command_interface_[index].get().set_value(state_desired.positions[index]);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         has_position_command_interface_
make[2]: *** [CMakeFiles/ur_controllers.dir/build.make:63: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/ur_controllers.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

MGBla, 2021-06-22:
Currently the scaled trajectory controller requires an older version of the ros2_controllers since there was a breaking change in their last update. This issue is currently addressed in #121

nbbrooks, 2021-07-01:
@sanatmharolkar PR 121 became https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/133 which is now merged in. Could you check if this resolves your issue?

nbbrooks, 2021-07-13:
@sanatmharolkar please reopen if this does not resolve the issue.

closed 2021-07-13 by nbbrooks