UniversalRobots/Universal_Robots_ROS2_Driver #134: Multiple robots ports

amadeuszsz, 2021-06-25:
Running two separate [nodes](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_bringup/launch/ur_control.launch.py#L227-L235) with different `robot_description` & `robot_controllers` causes ports complications. Despite changing [ports](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_description/urdf/ur.ros2_control.xacro#L24-L25), the trajectory_port remains hardcoded, isn't configurable within [hardware interface](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/42aa2ce518884a53b5cd07a16e14bc1e8f8ea4b8/ur_robot_driver/src/hardware_interface.cpp#L238-L255).

```
[ros2_control_node-1]   what():  Failed to bind socket for port 50003 to address. Reason: Address already in use: Address already in use
[ERROR] [ros2_control_node-1]: process has died 
```

I didn't see any suitable param in a package for trajectory_port within config files. Seems to be a new feature in [driver library](https://github.com/UniversalRobots/Universal_Robots_Client_Library/commit/ca8fc298f6cd38a970dc73b0df3edea0fda24324). Is it possible to run provided stack with two arms?

nbbrooks, 2021-08-12:
Will be addressed in port of https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/431

fmauch, 2022-06-27:
Yes, this is related to #420

closed 2024-05-29 by VinDp