UniversalRobots/Universal_Robots_ROS2_Driver #135: Custom URDF File to use with Moveit2

ppascotto, 2021-06-29:
I am trying to run the driver with moveit and a custom urdf description. I am using a ur10e, with other components (a frame the robot is attached to and cart) in the urdf file. On the readme, there is a mention of testing launch files and descriptions. I believe moveit2 does not have a setup assistant yet, how were the examples with the robot alone set up?

livanov93, 2021-07-01:
@ppascotto You are right, the moveit2 setup assistant is not ready yet, but it is on its way to became available in the future releases of Moveit2. 
However, you can easily follow [this moveit launch](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/main/ur_moveit_config) file which starts the robot with moveit2. You have to follow the existing instructions [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#example-commands-for-testing-the-driver) under the point _To test the driver with the example MoveIt-setup, first start the controllers then start MoveIt. (This requires a vcs import of MoveIt packages)_ .
For this it will be helpful to also check out [ur_moveit_config](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/main/ur_moveit_config) subpackage. Here you can find an examples of .srdf, .yaml and .rviz files that are used with basic configurations of ur robots with combination with moveit2 and ros2_control. If you take a closer look, the files are actually quite the same as in the ROS(1) version that are generated by setup_assistant. 
If you don't want to write some of the moveit2-required files manually for your specific robot ( it si not too complicated ) you can generate these files with setup_assistant from ROS(1) version and then just adapt the launch file which is referencing .srdf, .yaml and .rviz files.

livanov93, 2021-07-13:
@ppascotto Here is the [issue](https://github.com/ros-planning/moveit2/issues/313) you can follow for further information. Feel free to continue collaboration there.

closed 2021-07-13 by livanov93