UniversalRobots/Universal_Robots_ROS2_Driver #138: Issue with joint_trajectory_controller

freesolo92, 2021-07-05:
Hello guys,

I'm new in the ROS world and started with ROS2 (foxy) for my study work. We have some UR Cobots in our robot lab and I want to drive the robots via the ROS2 Driver. 
I use ROS2 in a Docker container because my host system is a ubuntu 18.04. I followed the setup description in the readme file. After some adaptions in the package.xml, I was able to compile the driver with colcon. But after a minute I got an error output:
______________________________________________________________________________________________________
--- stderr: joint_trajectory_controller
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findangles.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "angles", but
  CMake did not find one.

  Could not find a package configuration file provided by "angles" with any
  of the following names:

    anglesConfig.cmake
    angles-config.cmake

  Add the installation prefix of "angles" to CMAKE_PREFIX_PATH or set
  "angles_DIR" to a directory containing one of the above files.  If "angles"
  provides a separate development package or SDK, be sure it has been
  installed.

---
Failed   <<< joint_trajectory_controller [3.38s, exited with code 1]
Aborted  <<< forward_command_controller [8.25s]
Aborted  <<< diff_drive_controller [8.37s]
Aborted  <<< joint_state_broadcaster [20.0s]

Summary: 16 packages finished [1min 2s]
  1 package failed: joint_trajectory_controller
  3 packages aborted: diff_drive_controller forward_command_controller joint_state_broadcaster
  5 packages had stderr output: diff_drive_controller forward_command_controller joint_state_broadcaster joint_trajectory_controller ros2_control_test_nodes
  13 packages not processed
__________________________________________________________________________________________________________________

Maybe someone can give me a hint where to look for this issue and for a solution. I don't know if I did some wrong steps but the setup guide looked very easy.

Greetings,
Marcel

destogl, 2021-07-06:
Hi Marcel,

did you forgot to execute this line during setup:
```
rosdep install --ignore-src --from-paths src -y -r
```

This should automatically install all dependecies.

freesolo92, 2021-07-15:
Hi,

rosdep is already installed and the command `rosdep update` also works fine.
Maybe are there any redudant dependencies? With the trajectory controller from ros and the one from universal robots?

Greets.

livanov93, 2021-07-17:
@freesolo92 
The above-mentioned command @destogl  has indicated should install all the necessary dependencies needed to build your UR ros2 driver workspace. Please, follow the documentation in the [README](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#getting-started) carefully and don't skip steps. 

Cheers!

destogl, 2021-08-10:
@freesolo92 did you manage to resolve this?

freesolo92, 2021-08-11:
Sorry for the delayed response. 
I did not work further on this topic. I switched to ROS1 (Melodic Dist) and this ROS Driver works well.
I guess the problem is solved by installing all ros dependecies like mentioned above.

You can close this issue. Thanks for this fast support!

closed 2021-08-11 by freesolo92

Pomorondza, 2022-03-14:
After successfully installing all dependencies with the proposed command above, I still have this issue. Could anyone assist me with this?