UniversalRobots/Universal_Robots_ROS2_Driver #140: could not determine a constructor for the tag '!degrees'

NebuDa, 2021-07-08:
Hi,

I have tried to install the ROS2 Driver with this Howto: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver

At the Driver Testing I got the following Error:

```
flo@DESKTOP-L4I9CIG:~$ ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/flo/.ros/log/2021-07-08-13-15-59-881791-DESKTOP-L4I9CIG-138
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure('executed command failed. Command: /opt/ros/galactic/bin/xacro /home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/urdf/ur.urdf.xacro robot_ip:=yyy.yyy.yyy.yyy joint_limit_params:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/joint_limits.yaml kinematics_params:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/default_kinematics.yaml physical_params:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/physical_parameters.yaml visual_params:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/visual_parameters.yaml safety_limits:=true safety_pos_margin:=0.15 safety_k_position:=20 name:=ur5e script_filename:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/ros_control.urscript input_recipe_filename:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/rtde_input_recipe.txt output_recipe_filename:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/rtde_output_recipe.txt prefix:="" use_fake_hardware:=true fake_sensor_commands:=false \nCaptured stderr output: error: could not determine a constructor for the tag \'!degrees\'\n  in "/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/joint_limits.yaml", line 18, column 19 \nwhen evaluating expression \'load_yaml(joint_limits_parameters_file)\' \nwhen evaluating expression \'__joint_limit_parameters[\'joint_limits\']\' \nwhen evaluating expression \'__limits[\'shoulder_pan\'][\'min_position\']\'\nwhen instantiating macro: read_model_data (/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/urdf/inc/ur_common.xacro)\ninstantiated from: ur_robot (/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/urdf/ur_macro.xacro)\nin file: /home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/urdf/ur.urdf.xacro\n')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/actions/node.py", line 465, in execute
    self._perform_substitutions(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/actions/node.py", line 420, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 151, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 72, in evaluate_parameter_dict
    evaluated_value = perform_substitutions(context, list(value))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_context.py", line 197, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/substitutions/command.py", line 119, in perform
    raise SubstitutionFailure(on_error_message)
launch.substitutions.substitution_failure.SubstitutionFailure: executed command failed. Command: /opt/ros/galactic/bin/xacro /home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/urdf/ur.urdf.xacro robot_ip:=yyy.yyy.yyy.yyy joint_limit_params:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/joint_limits.yaml kinematics_params:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/default_kinematics.yaml physical_params:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/physical_parameters.yaml visual_params:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/visual_parameters.yaml safety_limits:=true safety_pos_margin:=0.15 safety_k_position:=20 name:=ur5e script_filename:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/ros_control.urscript input_recipe_filename:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/rtde_input_recipe.txt output_recipe_filename:=/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/rtde_output_recipe.txt prefix:="" use_fake_hardware:=true fake_sensor_commands:=false
Captured stderr output: error: could not determine a constructor for the tag '!degrees'
  in "/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/config/ur5e/joint_limits.yaml", line 18, column 19
when evaluating expression 'load_yaml(joint_limits_parameters_file)'
when evaluating expression '__joint_limit_parameters['joint_limits']'
when evaluating expression '__limits['shoulder_pan']['min_position']'
when instantiating macro: read_model_data (/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/urdf/inc/ur_common.xacro)
instantiated from: ur_robot (/home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/urdf/ur_macro.xacro)
in file: /home/flo/workspace/ros_ws_foxy_ur_driver/install/ur_description/share/ur_description/urdf/ur.urdf.xacro
```

I got the same Error when I try to connect to a URSim in VMWare. 

Thank you in advance.

Best regards
Florian

NebuDa, 2021-07-13:
@destogl thanks for linking. I am eagerly waiting for a solution.

destogl, 2021-07-13:
@NebuDa for what I can tell, this is a Galactic-related issue. something has changed there in xacro parsing or similar, which is breaking this. If you need this soon I propose to use Foxy.

NebuDa, 2021-08-20:
Changed ROS2 to Foxy and now it works. Should I close the Issue, or let it open for future Galactic integration?

destogl, 2021-09-16:
solved in #166

closed 2021-09-16 by destogl