UniversalRobots/Universal_Robots_ROS2_Driver #145: use move_group_interface ur5

studentWUR, 2021-07-27:
I am having some problems using the moveit group interface. I have succesfully installed moveit and the universal robot ros driver. Following line work fine 
`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false` 
`ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur5e robot_ip:="xxx.xxx" use_fake_hardware:=true launch_rviz:=true`

Now I wanted to use the move_group_interface for the ur5 arm instead of panda arm ([move_group_interface_tutorial](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/move_group_interface/launch/move_group_interface_tutorial.launch.py).cpp). The changed I made can be found below:
`

    robot_description_config = load_file(
        "ur_description", "urdf/ur.urdf.xacro"
    )

    robot_description = {"robot_description": robot_description_config}

    robot_description_semantic_config = load_file(
        "ur_moveit_config", "srdf/ur.srdf.xacro"
    )
    robot_description_semantic = {
        "robot_description_semantic": robot_description_semantic_config
    }
    
    kinematics_yaml = load_yaml(
        "ur_description", "config/default_kinematics.yaml"
    )
`
However, I get several error when launching the tutorial:

`ros2 launch moveit2_tutorials move_group_interface_tutorial.launch.py`
Anybody that successfully used the move group interface for a ur5e?

nbbrooks, 2021-08-03:
Could you post the error text you receive when running this?

studentWUR, 2021-08-05:
@nbbrooks 

> One thing that stands out is config/default_kinematics.yaml should be `config/ur5/default_kinematics.yaml"

Thank you that suggestion solved one error :+1: 
I am now able to start the Xterm, but after then i got the following error in Xterm:

```
Parsing robot urdf xml string
Error: no link elements found in urdf file at line 179 in /tmp/binarydeb/ros-foxy-urdfdom-2,3,3/urdf_parser/src/model.cpp
[INFO] [16828144719.16] [moveit_rdf_loader, rdf_loader]: unable to parse urdf from paameter robot_description
[FATAL] [16828144719.16] [move_group_interface]: unable to construct robot model pleakse make sure all needed information is on the parameter server.
terminate called after throwing an instane of 'std:runtime_error'
```
Do you have any suggestions?

destogl, 2021-08-10:
@studentWUR it seems that you are trying to start "move_group" tutorial. This should not be necessary here, If you start `ur_moveit.launch.py` from `ur_bringup` package you should get configured move group for the robots from this package. Also check the arguments of the launch file to select a right robot-type.

studentWUR, 2021-08-11:
@destogl I know that I need to use the ur_moveit.launch.py. And I am indeed able to control the robot using the motionplanner in Rviz. But then in the next step I want to send the robot to several Cartesian coordinates from a c++ script using the move group. At that point, I used the tutorial as an example, to send the UR5 to a specific location using the move_group_interace. Unfortunately, this is where I got stuck. I always used move_group with python in ros1. But as you know most of the moveit2 packages only work with c++ and as a result I am struggling to get the robot working. Do you have any suggestions or scripts available, how I could move the robot to a specific Cartesian Coordinate system.

JoNeckel, 2021-11-15:
Hello @studentWUR did you find any solution? I have the same problem

JoNeckel, 2021-11-15:
@destogl here https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/145#issuecomment-896043739 you mentioned to check the arguments of the launch file to select a right robot-type. Can you tell me how to select the right type?

```
def generate_launch_description():
    # planning_context
    robot_description_config = load_file(
        "ur_description", "urdf/ur.urdf.xacro" 
    )
    robot_description = {"robot_description": robot_description_config}
```
I think the problem lies in the "ur.urdf.xacro" without telling which ur type to use, but I have no idea how to select the ur-type in that launch file?

when I try to launch the move_group_interface_tutorial, I get this output:
```
jo@myubuntu:~/workspace/ros_ur_driver$ ros2 launch moveit2_tutorials move_group_interface_tutorial.launch.py
[INFO] [launch]: All log files can be found below /home/jo/.ros/log/2021-11-15-11-04-48-521394-myubuntu-180205
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group_interface_tutorial-1]: process started with pid [180207]
[move_group_interface_tutorial-1] [WARN] [1636970688.829893248] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[move_group_interface_tutorial-1] Parsing robot urdf xml string.
[move_group_interface_tutorial-1] Error:   No link elements found in urdf file
[move_group_interface_tutorial-1]          at line 179 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/model.cpp
[move_group_interface_tutorial-1] [INFO] [1636970688.835605998] [moveit_rdf_loader.rdf_loader]: Unable to parse URDF from parameter: 'robot_description'
[move_group_interface_tutorial-1] [FATAL] [1636970688.835666094] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.
[move_group_interface_tutorial-1] terminate called after throwing an instance of 'std::runtime_error'
[move_group_interface_tutorial-1]   what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.
[ERROR] [move_group_interface_tutorial-1]: process has died [pid 180207, exit code -6, cmd '/home/jo/workspace/ros_ur_driver/install/moveit2_tutorials/lib/moveit2_tutorials/move_group_interface_tutorial --ros-args -r __node:=move_group_interface_tutorial --params-file /tmp/launch_params_qy9q18tq --params-file /tmp/launch_params_wbvr2ym8 --params-file /tmp/launch_params_9budfden'].
```
when I check the urdf, I get following output:
```
jo@myubuntu:~/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_description/urdf$ check_urdf ur.urdf.xacro
Error:   No link elements found in urdf file
         at line 179 in /tmp/binarydeb/ros-foxy-urdfdom-2.3.3/urdf_parser/src/model.cpp
ERROR: Model Parsing the xml failed
```
which seems to be the same problem

destogl, 2022-02-28:
Hi @JoNeckel you should use `ur_type` argument. This is now documented very well. 

I will close this issue because it seems obsolete. Please update us if it is still an issue.

closed 2022-02-28 by destogl