UniversalRobots/Universal_Robots_ROS2_Driver #147: unable to build moveit on foxy: resource_retriever/retriever.hpp: No such file or directory

odbee, 2021-08-02:
hey, this is the error im getting when trying to build moveit. does anyone have a clue why this is happening? im still not sure if its a moveit issue or from the guide.

```zsh
Starting >>> control_msgs
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> controller_manager_msgs
Starting >>> moveit_common
Starting >>> ros2_control_test_assets
Starting >>> warehouse_ros
Finished <<< moveit_common [0.60s]
Starting >>> moveit_resources_panda_description
Finished <<< ros2_control_test_assets [0.66s]
Starting >>> moveit_resources_fanuc_description
Finished <<< warehouse_ros [0.94s]
Starting >>> moveit_resources_pr2_description
Finished <<< srdfdom [1.06s]
Starting >>> warehouse_ros_mongo
--- stderr: geometric_shapes
/home/ed/workspace/ros_ws_foxy_ur_driver2/src/geometric_shapes/src/mesh_operations.cpp:41:10: fatal error: resource_retriever/retriever.hpp: No such file or directory
   41 | #include <resource_retriever/retriever.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/geometric_shapes.dir/build.make:102: CMakeFiles/geometric_shapes.dir/src/mesh_operations.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/geometric_shapes.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< geometric_shapes [1.14s, exited with code 2]
Aborted  <<< moveit_resources_panda_description [0.50s]
Aborted  <<< moveit_resources_fanuc_description [0.41s]
Aborted  <<< warehouse_ros_mongo [0.06s]
Aborted  <<< moveit_resources_pr2_description [0.13s]
Aborted  <<< moveit_msgs [1.29s]
Aborted  <<< controller_manager_msgs [1.54s]
Aborted  <<< control_msgs [1.64s]

Summary: 4 packages finished [2.18s]
  1 package failed: geometric_shapes
  7 packages aborted: control_msgs controller_manager_msgs moveit_msgs moveit_resources_fanuc_description moveit_resources_panda_description moveit_resources_pr2_description warehouse_ros_mongo
  1 package had stderr output: geometric_shapes
  55 packages not processed
```

nbbrooks, 2021-08-05:
For now, you will need to build geometric_shapes from source until the next debian is released.

``
cd $COLCON_WS/src
git clone git@github.com:ros-planning/geometric_shapes.git -b foxy
```

nbbrooks, 2021-08-05:
Yes, there is currently an issue with resource retriever. For now, using the foxy branches of moveit2 and geometric_shapes should fix this while resource retriever renaming issue is addressed.

nbbrooks, 2021-08-05:
I will be updating the MoveIt_Support.repos to target the new moveit2 foxy branch, which has the revision to the moveit2.repos file to pull in the foxy branch of geometric_shapes. For now, you can resolve the problem by checking out the foxy branch of moveit2 and geometric_shapes.

NebuDa, 2021-08-18:
I had the same error and just changed in the mesh_operations.cpp Line 41 to "retriever.h". Then it worked for me.
Just as a quick workaround, @odbee .

cardboardcode, 2021-08-31:
Faced the same error. 

The workaround suggested by @NebuDa worked for me as well. 

## Solution
Replace **line 41** of `src/geometric_shapes/src/mesh_operations.cpp` from
```cpp
#include <resource_retriever/retriever.hpp>
```
To:
```cpp
#include <resource_retriever/retriever.h>
```

nbbrooks, 2021-09-23:
Repos files have all been updated!
- For use with foxy, checkout and follow instructions on `foxy` branch
- For use with galactic or rolling, checkout and follow instructions on `main` branch
- In the future, galactic will be split out as upstream API breaking changes are added to rolling

Please reopen if you run into issues using the latest repos files.

closed 2021-09-23 by nbbrooks