UniversalRobots/Universal_Robots_ROS2_Driver #150: Cannot move UR Sim

NebuDa, 2021-08-23:
Hi,
I am running ROS2 Foxy with the latest driver and latest UR Sim in VMWare. In the UR Sim I have installed and started the latest UR Cap "External Control".
I will get this running with the simulator before asking our partner for accessing the real UR10e.

I am able to see the moving robot in RViz, when moving within polyscope, but the execute Button in RViz MoveIt does not move the simulated UR10e. 
I use the following two Launch files:

```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.147.128 launch_rviz:=false
ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur10e robot_ip:="192.168.147.128" launch_rviz:=true
```

RViz and the Control Windows shows up this:

```
[ros2_control_node-1] [INFO] [1629709544.538141700] [joint_trajectory_controller]: Received new action goal
[ros2_control_node-1] [INFO] [1629709544.538233800] [joint_trajectory_controller]: Accepted new action goal
[ros2_control_node-1] [INFO] [1629709550.427761400] [joint_trajectory_controller]: Goal reached, success!
```

```
[rviz2-3] [INFO] [1629709544.836058300] [move_group_interface]: Execute request accepted
[move_group-1] [INFO] [1629709550.467701300] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Controller joint_trajectory_controller successfully finished
[move_group-1] [INFO] [1629709550.475229900] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status SUCCEEDED ...
[move_group-1] [INFO] [1629709550.475430400] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: SUCCEEDED
[rviz2-3] [INFO] [1629709550.848811500] [move_group_interface]: Execute request success!
```

UR Sim shows up this in Log: 

[image]

I have checked the network accessability and sending Joint information from Simulator to RViz works great, but I cannot send a trajectory to the Robot. Is there a problem with the Simulator or am I doing anything wrong?

If you need any further information just ask, I am very new to ROS2.
Thank you very much!

livanov93, 2021-08-24:
There is a working dockerized example [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/144). It is still under review but it is working and quite easy to follow.
Is your URcap for External Control playing for sure? Without that you can't stream trajectory to the simulator?
Please send the screenshot of the main URsim window.

NebuDa, 2021-08-25:
First two images of the Installation and Program Tab and last the selection for starting the External Control. I pushed the lower "Play from Selection" Button. 
[image]

[image]

[image]

Not sure if it showas anything that could help. 

The Docker is a nice idea, but I have to find out how to use it. I will check your readme.

NebuDa, 2021-09-06:
Hey, the problem is still there and I have found a new thing. When I start, or stop and start again the URCap in the URSimulator there comes this Log on the console where I have started: ```ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.40.128 launch_rviz:=false```

```[ros2_control_node-1] [INFO] [1630915962.376617700] [UR_Client_Library]: Connection to reverse interface dropped.
[ros2_control_node-1] [INFO] [1630915962.378616200] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
[ros2_control_node-1] [INFO] [1630915962.403155800] [UR_Client_Library]: Connection to reverse interface dropped.
[ros2_control_node-1] [INFO] [1630915962.405899300] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
[ros2_control_node-1] [INFO] [1630915962.431150300] [UR_Client_Library]: Connection to reverse interface dropped.
[ros2_control_node-1] [INFO] [1630915962.433475500] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
[ros2_control_node-1] [INFO] [1630915962.458400000] [UR_Client_Library]: Connection to reverse interface dropped.
[ros2_control_node-1] [INFO] [1630915962.461244600] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
[ros2_control_node-1] [INFO] [1630915962.490472300] [UR_Client_Library]: Connection to reverse interface dropped.
[ros2_control_node-1] [INFO] [1630915962.493090800] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
[ros2_control_node-1] [WARN] [1630915962.525849800] [UR_Client_Library]: Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.
[ros2_control_node-1] [INFO] [1630915962.525915900] [UR_Client_Library]: Connection to reverse interface dropped.
```

The first lines appears much oftener.

I thing here is the problem. Has anyone else seen this issue before?

livanov93, 2021-09-13:
It looks like you are trying to connect to the RTDE client multiple times. Make sure that your nodes and processes that communicate with ursim are shutdown before starting the nodes again.

NebuDa, 2021-09-14:
Thanks for the answers. 

@livanov93 I am just launching the two Launch Commands from the first Post
```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.147.128 launch_rviz:=false
ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur10e robot_ip:="192.168.147.128" launch_rviz:=true
```
Then the error appears instantly. 

Installation / Fieldbus / EtherNet/IP is disabled @urrsk .

deepparikh, 2021-09-17:
> Hey, the problem is still there and I have found a new thing. When I start, or stop and start again the URCap in the URSimulator there comes this Log on the console where I have started: `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=192.168.40.128 launch_rviz:=false`
> 
> ```
> [ros2_control_node-1] [INFO] [1630915962.378616200] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
> [ros2_control_node-1] [INFO] [1630915962.403155800] [UR_Client_Library]: Connection to reverse interface dropped.
> [ros2_control_node-1] [INFO] [1630915962.405899300] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
> [ros2_control_node-1] [INFO] [1630915962.431150300] [UR_Client_Library]: Connection to reverse interface dropped.
> [ros2_control_node-1] [INFO] [1630915962.433475500] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
> [ros2_control_node-1] [INFO] [1630915962.458400000] [UR_Client_Library]: Connection to reverse interface dropped.
> [ros2_control_node-1] [INFO] [1630915962.461244600] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
> [ros2_control_node-1] [INFO] [1630915962.490472300] [UR_Client_Library]: Connection to reverse interface dropped.
> [ros2_control_node-1] [INFO] [1630915962.493090800] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.
> [ros2_control_node-1] [WARN] [1630915962.525849800] [UR_Client_Library]: Connection attempt on port 50001 while maximum number of clients (1) is already connected. Closing connection.
> [ros2_control_node-1] [INFO] [1630915962.525915900] [UR_Client_Library]: Connection to reverse interface dropped.
> ```
> 
> The first lines appears much oftener.
> 
> I thing here is the problem. Has anyone else seen this issue before?

This is mostly due to the configuration you are using. The communication latency between the VM and ROS drivers results in this error. I faced a similar error and discussed it [here](https://forum.universal-robots.com/t/ursim-with-ros-package-ur-robot-driver/15560) in detail.

A solution that worked for me is to install URSim and ROS in a single VM / Linux machine. However, this is not easy and as discussed [here](https://forum.universal-robots.com/t/offline-simulator-e-series-ur-sim-for-linux-5-11-1-removes-all-installed-files/15384), the install script of URSim will remove all additional Linux packages from the system, hence be careful while installing the URSim. I would suggest, try with a VM image which you can easily break and replace.

NebuDa, 2021-09-21:
Thanks @deepparikh it was a timing Problem.
Now I have both the Simulator and ROS2 Foxy on Windows Subsystem for Linux 2 with Ubuntu-20.04 running and it works great. 

**This is a warning to all future readers:** Installation of the Linux UR Simulator leads to removing important ROS packages.

Thank you very much for your help. 
I'll close the Issue.

closed 2021-09-21 by NebuDa

JS-Yap, 2024-09-20:
Hi @NebuDa may I know how you fix the timing problem?

NebuDa, 2024-09-20:
As described I installed both in the same Environment.
I can barely remember, its a long time ago.

JS-Yap, 2024-09-20:
@NebuDa Thank you very much for your fast reply 😁