UniversalRobots/Universal_Robots_ROS2_Driver #163: Colcon not Building on Galactic with new

NebuDa, 2021-09-10:
I tried to build the driver now on Galactic installed from Debian Packages on Ubuntu 20.04 LTS because of trouble with building on Foxy. I show the output with a lot of Errors and hope anyone can help with it.

```
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> control_msgs
[0.509s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ros2_controllers' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.509s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/velocity_controllers' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.509s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ur_msgs' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.509s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ur_moveit_config' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.509s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ur_description' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.509s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ur_dashboard_msgs' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.509s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ur_bringup' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.510s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/transmission_interface' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.510s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control_demos' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.510s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control_demo_bringup' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.510s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/rrbot_description' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.510s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.510s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ros2controlcli' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.510s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control_test_nodes' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.510s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/controller_manager' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.511s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control_test_assets' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.511s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control_demo_hardware' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.511s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/position_controllers' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.511s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/joint_trajectory_controller' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.511s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/joint_state_broadcaster' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.511s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/imu_sensor_broadcaster' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.511s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/gripper_controllers' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.511s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/effort_controllers' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.512s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/forward_command_controller' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.512s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/force_torque_sensor_broadcaster' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.512s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/diff_drive_controller' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.512s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/controller_interface' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.512s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/hardware_interface' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.512s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/diffbot_description' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.512s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/controller_manager_msgs' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.512s] WARNING:colcon.colcon_ros.prefix_path.ament:The path '/home/flo/workspace/ros_ur_driver/install/control_msgs' in the environment variable AMENT_PREFIX_PATH doesn't exist
[0.513s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ros2_controllers' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.513s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/velocity_controllers' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.513s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ur_msgs' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.513s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ur_moveit_config' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.513s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ur_description' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.513s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ur_dashboard_msgs' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.514s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ur_client_library' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.514s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ur_bringup' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.514s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/transmission_interface' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.514s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control_demos' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.514s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control_demo_bringup' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.514s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/rrbot_description' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[0.514s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/flo/workspace/ros_ur_driver/install/ros2_control' in the environment variable CMAKE_PREFIX_PATH doesn't exist
[... cut ...]
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ur_controllers [11.9s, exited with code 2]

Summary: 32 packages finished [4min 43s]
  1 package failed: ur_controllers
  4 packages had stderr output: gripper_controllers joint_trajectory_controller ros2_control_demo_hardware ur_controllers
  1 package not processed
```

I have tried to rebuild it completely, but no change. I used the updated Installation Guide from this Repository main Branch (https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/142).

NebuDa, 2021-09-10:
Update: Same problem on clean new Rolling Installation.

livanov93, 2021-09-13:
For galactic and rolling use main branch, and there is separate branch for foxy.
[This PR](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/160/files) is fixing the build problems. It is oriented towards newest updates in ros-controls.

urrsk, 2021-09-14:
> For galactic and rolling use main branch, and there is separate branch for foxy.

@livanov93 I will suggest that we add a line to the 'Get Started' section mentioning the separate branch for foxy

NebuDa, 2021-09-14:
Have used Main Branch for tries with Galactic and Rolling. Haven't verified it again. It works now on Foxy, surlely built with Foxy Branch, with the old Issue https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/150 (cannot move Ur Sim with MoveIt!, but get Robot Joint Information in RViz).

Feel free to close the issue, if you see no trouble with building the main branch.

Thank you very much!

livanov93, 2021-09-20:
> > For galactic and rolling use main branch, and there is separate branch for foxy.
> 
> @livanov93 I will suggest that we add a line to the 'Get Started' section mentioning the separate branch for foxy

[The build status table](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#build-status) gives all the information needed. 

@NebuDa We will leave it opened for now, but the build status table should give the realistic view of the current state of the repo on multiple ros distros. The thing is that many ROS2 packages, especially [ros-controls](https://github.com/ros-controls) are being actively developed, so the ci and building procedures can fail quite often while we try to keep up with the changes on our main dependencies.