UniversalRobots/Universal_Robots_ROS2_Driver #167: Can't test "test_moveit_servo_velocity_controller.launch.py" successfully

Abelpy1, 2021-09-15:
Hello, recently I tried to test "test_moveit_servo_velocity_controller.launch.py", and my solutions to test this launch file are as follows:

1. Firstly, I just change the "robot_controller" in "ur_bringup/launch/ur_control.launch.py" from "joint_trajectory_controller" to "forward_velocity_controller", and then i run `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false` to start the robot driver and controllers. 
2. After that, I just run `ros2 launch ur_moveit_config test_moveit_servo_velocity_controller.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=True` to test the velocity controller.

The result is that the robot didn't move in Rviz. And i just got a warning "Invalid frame ID "world" passed to canTransform argument target_frame - frame does not exist".

Can you tell me  how to test "test_moveit_servo_velocity_controller.launch.py" correctly? And if we can test "test_moveit_servo_velocity_controller.launch.py" successfully by using the "fake_hardware"? 

Thank you very much and hope for your reply.

Abelpy1, 2021-09-17:
### Further attempts
Under the same environment, I tried that:

1. `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true`
2. `ros2 launch ur_bringup test_forward_velocity_controller.launch.py`
The result is that the robot in Rviz didn't move. And I check the state message of all joints, found that the "velocity state" change always, but the "position state" didn't change at all.

livanov93, 2021-09-20:
> ### Further attempts
> 
> Under the same environment, I tried that:
> 
>     1. `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true`
> 
>     2. `ros2 launch ur_bringup test_forward_velocity_controller.launch.py`
> 
> 
> The result is that the robot in Rviz didn't move. And I check the state message of all joints, found that the "velocity state" change always, but the "position state" didn't change at all.
> ### Experiments on real robots
> 
> Finally, I replace "fake_hardware" with "real robots", and find that everything is OK. The robot can move.
> 
>     * So, I'm very confused that whether it's true that if we want to use velocity_controller to make the robot move, we should not use the fake hardware.

The problem is that [fake driver in ros2_control](https://github.com/ros-controls/ros2_control/blob/master/hardware_interface/src/fake_components/generic_system.cpp) does not yet support velocity commands, however tests should run with no problem on the real hardware. 
To test velocity controller in the simulation you can check t[his branch](https://github.com/livanov93/ros2_control/tree/livanov/simulate_disconnected_driver) which has not yet been prepared as PR into ros2_control, but it supports velocity mode.

closed 2021-10-04 by livanov93