UniversalRobots/Universal_Robots_ROS2_Driver #168: Parameter naming in joint_limits.yaml

christianlandgraf, 2021-09-16:
Hi,

when trying to test the Pilz Planner in ROS 2 with the UR, I noticed `joint_limits.yaml` in the ROS 2 driver specifies the limits with regard to a generic name (e.g. wrist_3, wrist_2 etc) instead of the actual joint name (e.g. wrist_3_joint, wrist_2_joint) in the urdf.

See [ROS 1](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur10e_moveit_config/config/joint_limits.yaml) and [ROS 2](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_description/config/ur10e/joint_limits.yaml)

Is there any reason for changing this?
Since the Pilz motion planner checks if velocity limits (and acceleration limits) exist for each joint in a planning group, it does not correctly load the parameters (e.g. [here](https://github.com/sjahr/moveit2/blob/pr-port_pilz_planner/moveit_planners/pilz_industrial_motion_planner/src/trajectory_generator_ptp.cpp#L79)).

I didn't encounter a problem somewhere else in MoveIt, so far, but theoretically it could appear in other parts as well?

destogl, 2021-09-23:
This seems rather as an issue with wrong configuration of joint to plan for. You have to specify them somewhere with the suffix `_joint`. Please try to check this.

@livanov93, @urrsk do you think we should change names as they were in ROS1?

livanov93, 2021-09-23:
@destogl The thing is that links that have been compared are not from the same packages. The [ROS1](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur10e_moveit_config/config/joint_limits.yaml) version that @christianlandgraf  has referenced is from `urXYZ_moveit_config` package and [the one ROS2 version referenced](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_description/config/ur10e/joint_limits.yaml) is in `ur_description` package.
If we take a look in [ROS1 version](https://github.com/fmauch/universal_robot/blob/calibration_devel/ur_description/config/ur5e/joint_limits.yaml) of `ur_description` it is obvious it has the same form (without `_joint` suffix).

The ones from the description package are used for propagating parameters to robot_description via yaml file (the final steps can be seen [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/f43f9f728f28f1823d6076df1367be5ce18548c7/ur_description/urdf/inc/ur_common.xacro#L40)) and the other one is used as parameters in moveit planning (see example [here](https://github.com/ros-planning/moveit2/blob/cee20d6213ec1d4d1db85ecbe879f23c60a5289b/moveit_demo_nodes/run_move_group/launch/run_move_group_interface.launch.py#L48)).

I propose following action(s):

- Add additional joint limits yaml file for moveit planning (e.g. [here](https://github.com/ros-planning/moveit2/blob/cee20d6213ec1d4d1db85ecbe879f23c60a5289b/moveit_demo_nodes/run_move_group/launch/run_move_group_interface.launch.py#L48)) in our [moveit launch file](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/f43f9f728f28f1823d6076df1367be5ce18548c7/ur_bringup/launch/ur_moveit.launch.py#L292) and [servo launch file](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/f43f9f728f28f1823d6076df1367be5ce18548c7/ur_moveit_config/launch/test_moveit_servo_velocity_controller.launch.py#L263).
- In the [ur_moveit_config/config](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/main/ur_moveit_config/config) add folder with multiple per robot type subfolders that will contain specific joint_limits.yaml file that will be used for planning

If we go immediately to change our `ur_description` joint limits yaml, we will have to change some of the core xacros in the same package which is  actually shared all over the ROS1 and ROS2 community at the moment.

urrsk, 2021-09-23:
In the past, people have limit the robot actually joint limits used for planning to help moveit. This might be the reason why the are named differently.

In addition, I can see that the angle limits for UR3's wrist 3 is set to ±2pi, though it is in reality unlimited.
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_description/config/ur3e/joint_limits.yaml

destogl, 2021-09-23:
> In addition, I can see that the angle limits for UR3's wrist 3 is set to ±2pi, though it is in reality unlimited.

We should remove this. Though this could be useful for MoveIt to reduce planning space. Sometimes algorithms "get lost" on infinite joints.

gavanderhoorn, 2021-09-23:
I'd suggest to carefully consider whether to change these kinds of things.

Especially joint limits have been configured over time to reduce problems with motion planners (MoveIt being one of them).

Infinite limits (or `continuous`) joints can be nice, but infinite planning spaces typically do not result in better plans. Users targetting applications such as screwing or drilling would probably not control that action using the driver, making the need to 'unlimit' wrist 3 even less.

I would attribute the discrepancy between joint names to a simple mistake. In the early stages of the ROS 2 driver, things have been copied from upstream packages without much discussion. The `.yaml` file being discussed was not intended to be used as joint limits for MoveIt, but instead only as input for the respective `.xacro` files. The `xacro:macro` didn't use the full joint names as it didn't need to.

For context: https://github.com/ros-industrial/universal_robot/pull/516, https://github.com/ros-industrial/universal_robot/pull/513, https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/198 and https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/223.