UniversalRobots/Universal_Robots_ROS2_Driver #172: Foxy Branch build issue: pls change repo-Files

NebuDa, 2021-09-21:
I had again trouble with building the Foxy Branch and, as before mentionend by @livanov93 in https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/163, the ROS2 Packages changes quick. Some Repositories have new Foxy Branches.

Please change in Foxy Branch the following repo Files to make it buildable again.

src/Universal_Robots_ROS2_Driver/MoveIt_Support.repos:

```
repositories:
  moveit2:
    type: git
    url: https://github.com/ros-planning/moveit2.git
    version: foxy
  srdfdom:
    type: git
    url: https://github.com/ros-planning/srdfdom.git
    version: ros2
  warehouse_ros_mongo:
    type: git
    url: https://github.com/ros-planning/warehouse_ros_mongo
    version: ros2

```
src/moveit2/moveit2.repos:

```
repositories:
  moveit_msgs:
    type: git
    url: https://github.com/ros-planning/moveit_msgs
    version: ros2
  moveit_resources:
    type: git
    url: https://github.com/ros-planning/moveit_resources
    version: ros2
  geometric_shapes:
    type: git
    url: https://github.com/ros-planning/geometric_shapes
    version: foxy
  srdfdom:
    type: git
    url: https://github.com/ros-planning/srdfdom
    version: ros2
  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control
    version: foxy
  ros2_controllers:
    type: git
    url: https://github.com/ros-controls/ros2_controllers
    version: foxy
  warehouse_ros:
    type: git
    url: https://github.com/ros-planning/warehouse_ros
    version: foxy
  warehouse_ros_mongo:
    type: git
    url: https://github.com/ros-planning/warehouse_ros_mongo
    version: ros2

```

src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos

```
repositories:
  control_msgs:
    type: git
    url: https://github.com/ros-controls/control_msgs.git
    version: foxy-devel
  ros2_control:
    type: git
    url: https://github.com/ros-controls/ros2_control.git
    version: foxy
  ros2_control_demos:
    type: git
    url: https://github.com/ros-controls/ros2_control_demos.git
    version: master
  ros2_controllers:
    type: git
    url: https://github.com/ros-controls/ros2_controllers.git
    version: foxy
  Universal_Robots_Client_Library:
    type: git
    url: https://github.com/UniversalRobots/Universal_Robots_Client_Library.git
    version: master
  ur_msgs:
    type: git
    url: https://github.com/destogl/ur_msgs.git
    version: ros2
```

I am very new to github, so if there is a better way to show you something like this, please tell me.

livanov93, 2021-09-21:
@NebuDa, we very appreciate any kind of help and the best way to make the contribution to the open-source community and for instance this repository is to open an issue and then a related PR (pull request). 
[Here](https://github.com/firstcontributions/first-contributions) is an easy-to-follow instructions to test your first (dummy) PR and then you can then easily replicate the steps to do the same on Universal_Robots_ROS2_Driver repository.
Thank you very much for heads-up!

destogl, 2021-09-23:
The files are adjusted. #180 will resolve the last piece of this issue.

livanov93, 2021-09-23:
Closing this one as dependencies that have foxy branch are set to it.

closed 2021-09-23 by livanov93