UniversalRobots/Universal_Robots_ROS2_Driver #174: Build status does not update correctly

livanov93, 2021-09-21:
[Build status in the README.md](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#build-status) does not update its flags and badges correctly.

urrsk, 2021-09-22:
A lot of the builds are red! In addition, the badge url for the semi and source build are the same for all the three..!

I have here added the branch argument to the url for foxy. But are the CI running in that branch?

ROS2 Distro | Foxy  | Galactic | Rolling
:---------: | :---: | :------: | :-----:
**Branch** | [foxy](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/foxy) | [main](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/main) | [main](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/main)
**Build Status** | [image] <br /> [image] <br /> [image] | [image] <br /> [image] <br /> [image] | [image] <br /> [image] <br /> [image]

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/actions/workflows/ci-build-source.yml/badge.svg?branch=foxy

livanov93, 2021-09-22:
@urrsk Yes, the CI is also running for `foxy` branch. We need a PR to tackle this. Currently the build status on the main branch does not update correctly, the foxy branch CIs are passing at the moment.

urrsk, 2021-09-22:
@livanov93 Okay, so maybe you can use my suggestion for show the status on the foxy branch

livanov93, 2021-09-22:
@urrsk The link you provided is more than half of the work that needs to be done. Thank you!

PR #176 (2021-09-22): [foxy] Fix ci status badges
Closes #174
PR #175 (2021-09-22): [main] Fix ci status badges
Closes #174
closed 2021-09-22 by destogl