UniversalRobots/Universal_Robots_ROS2_Driver #181: remote control and dashboard service control

mdragusu, 2021-09-23:
I am trying to use dashboard commands with ros2. The driver is working fine and connected to the UR5e but when I try to:
```ros2 service call  /dashboard_client/play std_srvs/srv/Trigger```
I get:
```waiting for service to become available...
requester: making request: std_srvs.srv.Trigger_Request()

response:
std_srvs.srv.Trigger_Response(success=False, message='Command is not allowed due to safety reasons, please switch robot to Remote Control mode and reconnect to port 29999')
```
Connection seems to be there 
dashboard 1843981            root   18u  IPv4 2848002      0t0  TCP <my_ip>:41800-><UR5_ip>:29999 (ESTABLISHED)

And the robot is in remote control
[image]

when trying to do the same with socket commands I get:
```
Received from server: Connected: Universal Robots Dashboard Server

Received from server: Command is not allowed due to safety reasons, please switch robot to Remote Control mode and reconnect to port 29999
```

Is there something that I am missing?

destogl, 2021-09-23:
@mdragusu did you exactly followed the instruction in ["Hardware Setup" section](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#hardware-setup)

mdragusu, 2021-09-23:
Yes those steps were followed, I installed the external control urcap file
 maybe this can help prove it?
```
ls ~/GUI/bundle/urcaps/com.fzi.externalcontrol-1.0.5.jar 
/root/GUI/bundle/urcaps/com.fzi.externalcontrol-1.0.5.jar
```
Also the external control under Robot program works well, since I can run the provided ros2 test samples and the robot moves.

urrsk, 2021-09-23:
> And the robot is in remote control
> [image]

@mdragusu Actually, when looking at the screen dump, I can see that you have allowed it to go into remote control. But you are still in local mode. You need to click in right upper corner and change it from local to remote.
Maybe also see: http://help.universal-robots.com/SW_5_11/UR10e/Content/from-tex/software_manual/g5_sections/hamburger_menu_g5/settings_en.htm

mdragusu, 2021-09-24:
That was it, thank you.
I missed that button.

closed 2021-09-24 by mdragusu