UniversalRobots/Universal_Robots_ROS2_Driver #183: Clock synchronization issues with MoveIt Move Group via RViz

AndyZe, 2021-09-24:
Hi,

I've launched things just fine per the README and everything seems to be working well:

```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur3 robot_ip:=192.168.1.102 use_fake_hardware:=false launch_rviz:=false

(start the URCap)

ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur3 robot_ip:="192.168.1.102" use_fake_hardware:=false launch_rviz:=true
```

The robot's position in RViz looks good and all the controllers are in `active` state. However, when clicking "Plan and Execute", I get this clock synch error:
```

[move_group-1] [INFO] [1632511268.103625867] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[rviz2-3] [INFO] [1632511268.403802188] [move_group_interface]: Execute request accepted
[move_group-1] [INFO] [1632511269.103766391] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[move_group-1] Check clock synchronization if your are running ROS across multiple machines!
```

This is with an older CB3 UR3 robot. It has the most recent UR external control program (v1.0.5 I think).

I am using ROS Rolling and the `main` branch of this repo.

livanov93, 2021-09-27:
Hi @AndyZe,

We are already aware of this problem regarding incorrect timestamping in the `ros2_control_node`. I have tested ros-controls/ros2_control#542 with URsim and it worked perfectly. To improve moveit planning we just have to address #168 so we define precisely planning space. It has already been discussed and will be fixed soon.
Thank you for the issue!

destogl, 2021-09-27:
ros-controls/ros2_control#542 is merged.

AndyZe, 2021-09-28:
I don't have a good opportunity to test right now. I'll assume it's fixed. Thanks!

closed 2021-09-28 by AndyZe