UniversalRobots/Universal_Robots_ROS2_Driver #192: UR10e continues moveing after reaching goal

Thijsons, 2021-10-01:
Hello,

We are working on controlling a UR10e via Moveit2. Right now we are trying to run the Moveit demo but the robot keeps moving after reaching its goal state.

System:
- Ros2 foxy
- Moveit2
- UR ros2 drivers, foxy branch
- externalcontrol-1.0.5.urcap is at version 1.0.5

Steps to reproduce:
-ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=false launch_rviz:=false

-start the external_control on the pendant

-ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur10e robot_ip:="xxx.xxx" use_fake_hardware:=false launch_rviz:=true

-Set some new goal in moveit

-Click plan and execute

Expected result:
robot moves to the goal state and stops

Real results:
robot moves to goal state and keeps on moving (doesn't stop) in the same direction. Protective stop is activated automatically. It gives the following error:  "Runtime error: The program used too much time without instructing the robot what to do. Consider using a Wait  or sync() to split program sequences that do not move the robot"

Output:
the console running 'ur_control.launch.py' outputs the following:
>[ros2_control_node-1] [INFO] [1633094423.012856543] [joint_trajectory_controller]: Received new action goal
>[ros2_control_node-1] [INFO] [1633094423.012913503] [joint_trajectory_controller]: Accepted new action goal
>[ros2_control_node-1] [INFO] [1633094423.672177221] [joint_trajectory_controller]: Goal reached, success!

once the Protective stop (or emergency stop) is used the program loses connection:
>[ros2_control_node-1] [INFO] [1633094430.342446341] [UR_Client_Library]: Connection to reverse interface dropped.

Meanwhile Moveit/Rviz confirms that the execution was a success.

Thijsons, 2021-10-04:
Seems like we managed to solve it by increasing the speed slider on the teach pendant. We had it down to 10% for testing, upping it to 50% seems to have solved the issue. Not quite sure why this happens.

gavanderhoorn, 2021-10-05:
This seems like it should remain open.

Taking the scaling into account on the ROS side is explicitly a feature of this driver -- or: it should be.

If this doesn't work, that would be a regression from the ROS 1 version.

Thijsons, 2021-10-05:
That's a good point, I will reopen the issue. Not quite sure how to add a 'bug' tag to an issue but hat might be appropriate.

gavanderhoorn, 2021-10-05:
I would not do that unless it's been confirmed.

livanov93, 2021-10-05:
@Thijsons Can you tell me the `/joint_states` publish frequency? You can check that with `ros2 topic hz /joint_states`. Also answer me some other questions:

- Was urcap playing all the time while trajectory was executed?
- How long has the robot been moving until going into EMS?
- Which controller was active? Test this with `ros2 control list_controllers` and you will see the list of active controllers.
- Have you tested the driver on your robot with other ROS2 distros (e.g. galactic, rolling)?

Thank you for the issue! 
@gavanderhoorn Thanks for your timeliness!

Thijsons, 2021-10-05:
**joint_states publish frequency:** 
500hz on average, this doesn't change when we change the speed slider on the pendant.

**Was urcap playing all the time while trajectory was executed?** 
urcap seems to keep playing until the protective stop is automatically activated, then it stops.

**How long has the robot been moving until going into EMS?** 
This depends, can be 1-2 seconds. Sometimes it last for long enough that we have to manually activate the emergency stop to stop it from hitting any obstacles.
  
**Which controller was active? Test this with ros2 control list_controllers and you will see the list of active controllers.**
		io_and_status_controller[ur_controllers/GPIOController] active    
		speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active    
		joint_trajectory_controller[joint_trajectory_controller/JointTrajectoryController] active    
		joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
		force_torque_sensor_broadcaster[ur_controllers/ForceTorqueStateBroadcaster] active   
    
**Have you tested the driver on your robot with other ROS2 distros (e.g. galactic, rolling)?**
    		We have not, but we have tried ros2 foxy on multiple machines and even with a clean install.

livanov93, 2021-10-20:
@Thijsons You have to change [this](https://github.com/ros-planning/moveit2/blob/348146fb9e77510498f0151523c144ac9f2f80a3/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L200) parameter because it is set to `true` in the [init phase](https://github.com/ros-planning/moveit2/blob/348146fb9e77510498f0151523c144ac9f2f80a3/moveit_ros/planning/trajectory_execution_manager/src/trajectory_execution_manager.cpp#L96).

You can do this by following actions:

- Change the launch file `ur_moveit.launch.py` [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/ee85a93395d7a67aca516169a69db9874adfa90a/ur_bringup/launch/ur_moveit.launch.py#L169) from

```
moveit_controllers = {
        "moveit_simple_controller_manager": controllers_yaml,
        "moveit_controller_manager": "moveit_simple_controller_manager/MoveItSimpleControllerManager",
    }
```
to
```
moveit_controllers = {
        "moveit_simple_controller_manager": controllers_yaml,
        "moveit_controller_manager": "moveit_simple_controller_manager/MoveItSimpleControllerManager",
        "trajectory_execution.execution_duration_monitoring": False,
    }
```

- Call the service to set this parameter by:
```
ros2 service call /moveit_simple_controller_manager/set_parameters rcl_interfaces/srv/SetParameters "{parameters: [{name: "trajectory_execution.execution_duration_monitoring", value: {type: 1, bool_value: False}}]}"

```

gavanderhoorn, 2021-10-21:
@Thijsons wrote:
> The controller seems to notice the UR10e lagging behind and cancels the execution.

just to clarify: this is not the `ros_control` controller, but MoveIt.

That's why @livanov93 suggests to disable *trajectory execution monitoring*. MoveIt is currently not aware of this scaling, so it doesn't know that the fact the robot taking longer than expected is "normal".