UniversalRobots/Universal_Robots_ROS2_Driver #193: Unable to locate definitions for rosdep install using latest Foxy release

3ngineerit, 2021-10-03:
I recently installed the latest release of Foxy ROS2 (via debian packages) on my Ubuntu 20.04 VM on my system. My end goal is to get my system configured to start using the UR5e robots I have here in the lab. 

https://github.com/ros2/ros2/releases/tag/release-foxy-20210902

Everything performed nominally on that installation with validation.

When I came to performing the installation steps for this ROS2 driver, I followed the procedures and ensured that I cloned the foxy branch of the repo. However, when I run the rosdep install step of the installation steps, I get the following errors:

user@ubuntu:~/workspace/ros_ur_driver$ rosdep install --ignore-src --from-paths src -y -r
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
ros2controlcli: Cannot locate rosdep definition for [ament_xmllint]
ros2_control_demos: Cannot locate rosdep definition for [ament_cmake]
controller_interface: Cannot locate rosdep definition for [ament_lint_common]
force_torque_sensor_broadcaster: Cannot locate rosdep definition for [ament_cmake_gmock]
imu_sensor_broadcaster: Cannot locate rosdep definition for [ament_lint_common]
controller_manager: Cannot locate rosdep definition for [ament_lint_common]
forward_command_controller: Cannot locate rosdep definition for [ament_cmake_gmock]
diffbot_description: Cannot locate rosdep definition for [rviz2]
controller_manager_msgs: Cannot locate rosdep definition for [ament_lint_common]
ros2_control: Cannot locate rosdep definition for [ament_cmake]
ros2_control_demo_hardware: Cannot locate rosdep definition for [ament_cmake_gtest]
hardware_interface: Cannot locate rosdep definition for [ament_lint_common]
diff_drive_controller: Cannot locate rosdep definition for [ament_cmake_gmock]
effort_controllers: Cannot locate rosdep definition for [ament_cmake_gmock]
ur_controllers: Cannot locate rosdep definition for [std_srvs]
ros2_control_demo_bringup: Cannot locate rosdep definition for [xacro]
ur_description: Cannot locate rosdep definition for [xacro]
ros2_control_test_nodes: Cannot locate rosdep definition for [trajectory_msgs]
joint_state_broadcaster: Cannot locate rosdep definition for [rclcpp]
control_msgs: Cannot locate rosdep definition for [ament_lint_common]
velocity_controllers: Cannot locate rosdep definition for [ament_cmake_gmock]
ros2_control_test_assets: Cannot locate rosdep definition for [ament_lint_auto]
transmission_interface: Cannot locate rosdep definition for [ament_lint_common]
ur_robot_driver: Cannot locate rosdep definition for [launch_testing_ament_cmake]
ur_moveit_config: Cannot locate rosdep definition for [ament_cmake]
ur_dashboard_msgs: Cannot locate rosdep definition for [action_msgs]
position_controllers: Cannot locate rosdep definition for [ament_cmake_gmock]
gripper_controllers: Cannot locate rosdep definition for [ament_cmake_gmock]
joint_trajectory_controller: Cannot locate rosdep definition for [ament_cmake_gtest]
ur_bringup: Cannot locate rosdep definition for [xacro]
ros2_controllers: Cannot locate rosdep definition for [ament_cmake]
ur_msgs: Cannot locate rosdep definition for [ament_lint_common]
rrbot_description: Cannot locate rosdep definition for [rviz2]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
user@ubuntu:~/workspace/ros_ur_driver$ 

I could not find any particular method in resolving these errors. I tried a bunch of different methods proposed elsewhere, with no avail.

Perhaps this community might have a good resolution for this.

(I'm new to ROS, so I apologize if my issue statement comes off amateur)

3ngineerit, 2021-10-03:
Update: Ran the rosdep install command with the "--rosdistro=foxy" extension as recommended by the error code. It states that all rosdeps are installed. However, when I run the following colcon build command, I get conflicts with missing cmake packages (hence my concern about how to install the foxy version of this driver). 

erik@ubuntu:~/workspace/ros_ur_driver$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> control_msgs
Starting >>> ros2_control_test_assets
Starting >>> controller_manager_msgs
Starting >>> diffbot_description
--- stderr: control_msgs                                                                                  
CMake Error at CMakeLists.txt:13 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

  ament_cmakeConfig.cmake
  ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< control_msgs [0.16s, exited with code 1]
Aborted  <<< diffbot_description [0.12s]
Aborted  <<< controller_manager_msgs [0.13s]
Aborted  <<< ros2_control_test_assets [0.14s]

Summary: 0 packages finished [0.56s]
  1 package failed: control_msgs
  3 packages aborted: controller_manager_msgs diffbot_description ros2_control_test_assets
  4 packages had stderr output: control_msgs controller_manager_msgs diffbot_description ros2_control_test_assets
  30 packages not processed

Is there also a command workaround to perform the colcon build for the foxy build?

dennyvt, 2021-10-04:
I am also not a ROS pro, but most of the time this error (Could not find a package configuration file provided by "ament_cmake" with
any of the following names:

ament_cmakeConfig.cmake
ament_cmake-config.cmake) 
 happens when you haven't sourced ROS. For every terminal you open make sure you have sourced it. For example: source/opt/ros/foxy/setup.bash 
Hope this will work for you. I have  the drivers also installed on foxy so it should work in some way :)

livanov93, 2021-10-04:
Before running `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release` try to exceute `source /opt/${ROS_DISTRO}/setup.bash` and then try the build command.

3ngineerit, 2021-10-04:
That seemed to work, thanks a ton for that.

I noticed the following wasn't installed when running the colcon build:

user@ubuntu:~/workspace/ros_ur_driver$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> control_msgs
Starting >>> ros2_control_test_assets
Starting >>> controller_manager_msgs
Starting >>> diffbot_description
Finished <<< ros2_control_test_assets [3.08s]                                                                                                    
Starting >>> ros2_control_test_nodes                                                                                                      
Finished <<< diffbot_description [4.15s]                                                                                                      
Starting >>> rrbot_description
Finished <<< ros2_control_test_nodes [11.2s]                                                                                                     
Starting >>> ur_client_library
Finished <<< rrbot_description [26.0s]                                                                              
Starting >>> ur_dashboard_msgs
[Processing: control_msgs, controller_manager_msgs, ur_client_library, ur_dashboard_msgs]                                       
Finished <<< controller_manager_msgs [1min 21s]                                                                                     
Starting >>> ur_msgs                                                                                                        
Finished <<< ur_client_library [1min 19s]                                                                                     
Starting >>> ur_description
Finished <<< ur_description [4.96s]                                                                                         
Starting >>> ur_moveit_config
Finished <<< ur_moveit_config [2.43s]                                                                                         
Finished <<< control_msgs [1min 41s]                                                                                       
Starting >>> hardware_interface
Finished <<< ur_dashboard_msgs [1min 30s]                                                                                                        
Finished <<< ur_msgs [56.3s]                                                                                 
Finished <<< hardware_interface [1min 5s]                                
Starting >>> controller_interface
Starting >>> ros2_control_demo_hardware
Starting >>> transmission_interface
--- stderr: ros2_control_demo_hardware                                                                                             
In file included from /home/erik/workspace/ros_ur_driver/src/ros2_control_demos/ros2_control_demo_hardware/src/diffbot_system.cpp:15:
/home/erik/workspace/ros_ur_driver/src/ros2_control_demos/ros2_control_demo_hardware/include/ros2_control_demo_hardware/diffbot_system.hpp:22:10: fatal error: hardware_interface/base_interface.hpp: No such file or directory
   22 | #include "hardware_interface/base_interface.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ros2_control_demo_hardware.dir/build.make:63: CMakeFiles/ros2_control_demo_hardware.dir/src/diffbot_system.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/erik/workspace/ros_ur_driver/src/ros2_control_demos/ros2_control_demo_hardware/src/rrbot_system_position_only.cpp:15:
/home/erik/workspace/ros_ur_driver/src/ros2_control_demos/ros2_control_demo_hardware/include/ros2_control_demo_hardware/rrbot_system_position_only.hpp:22:10: fatal error: hardware_interface/base_interface.hpp: No such file or directory
   22 | #include "hardware_interface/base_interface.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ros2_control_demo_hardware.dir/build.make:76: CMakeFiles/ros2_control_demo_hardware.dir/src/rrbot_system_position_only.cpp.o] Error 1
In file included from /home/erik/workspace/ros_ur_driver/src/ros2_control_demos/ros2_control_demo_hardware/src/rrbot_system_multi_interface.cpp:15:
/home/erik/workspace/ros_ur_driver/src/ros2_control_demos/ros2_control_demo_hardware/include/ros2_control_demo_hardware/rrbot_system_multi_interface.hpp:25:10: fatal error: hardware_interface/base_interface.hpp: No such file or directory
   25 | #include "hardware_interface/base_interface.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ros2_control_demo_hardware.dir/build.make:89: CMakeFiles/ros2_control_demo_hardware.dir/src/rrbot_system_multi_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/ros2_control_demo_hardware.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ros2_control_demo_hardware [5.03s, exited with code 2]
Aborted  <<< transmission_interface [26.3s]                                                                                             
Aborted  <<< controller_interface [31.5s]                                         
                                        
Summary: 12 packages finished [3min 18s]
  1 package failed: ros2_control_demo_hardware
  2 packages aborted: controller_interface transmission_interface
  2 packages had stderr output: ros2_control_demo_hardware transmission_interface
  19 packages not processeduser@ubuntu:~/workspace/ros_ur_driver$

livanov93, 2021-10-04:
@3ngineerit Which branch are you using? You should be on [foxy](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/foxy).

3ngineerit, 2021-10-04:
Oh whoops, that would have done it for sure.

It looks like the COLCON build functioned nominally. 

Thanks a ton!

One final question regarding the build (again sorry for my novice questions).

Trying to source the build after running all of the installation steps, I get the following error: 

I double checked and setup.bash does exist in the install/ directory under the path listed on the error. No errors were reported in the COLCON build. 

erik@ubuntu:~/workspace/ros_ur_driver$ source ~/workspace/ros_ur_driver/install/setup.bash 
not found: "/home/erik/install/controller_manager/share/controller_manager/local_setup.bash"
[connext_cmake_module] Warning: The location at which Connext was found when the workspace was built [[/opt/rti.com/rti_connext_dds-5.3.1]] does not point to a valid directory, and the NDDSHOME environment variable has not been set. Support for Connext will not be available.
erik@ubuntu:~/workspace/ros_ur_driver$

livanov93, 2021-10-04:
I think there is something wrong with your [DDS](https://docs.ros.org/en/foxy/Installation/DDS-Implementations.html). Has your ROS2-foxy installation passed without any problems? Usually the pseudo-steps are:

- install ROS(2)
- source /opt/${ROS_DISTRO}/setup.bash
- rosdep install ...
- colcon build ...
- source ~/ros_ws/install/setup.bash

3ngineerit, 2021-10-05:
It looks like what happened was there were conflicting installs of Foxy on my system. I went through and completely removed those installs and cleared the directories. After running through the build again, no issues were discovered.

Thanks everyone for the help on this!

closed 2021-10-05 by 3ngineerit