UniversalRobots/Universal_Robots_ROS2_Driver #195: MoveIt! Scene Objects are ignored or dissapear

NebuDa, 2021-10-05:
After starting the Driver and Moveit I can plan paths with MoveIt and the URSimulator, but when I try to add a scene object as obstacle, MoveIt ignores the Object. When I attach the Object to the base_link it's getting deleted.
I use the example with following commands:
```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=127.0.0.1 launch_rviz:=false
```
Then starting the URCap
```
ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur10e robot_ip:="127.0.0.1" launch_rviz:=true
```

I will show you some screenshots to show the problem:

[image]

Attaching Problem:
Klicking ok and then "Plan" in the Planning Tab let the Box_0 dissappear.

[image]

And here is the Log Output from ur_moveit.launch.py:

```
ur@NBK-18:~$ ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur10e robot_ip:=127.0.0.1 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/ur/.ros/log/2021-10-05-14-31-29-260094-NBK-18-11396
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [11402]
[INFO] [mongo_wrapper_ros.py-2]: process started with pid [11404]
[INFO] [rviz2-3]: process started with pid [11406]
[INFO] [static_transform_publisher-4]: process started with pid [11408]
[static_transform_publisher-4] [INFO] [1633437090.400657300] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'base_link'
[move_group-1] [WARN] [1633437090.460087900] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1633437090.461019900] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] Parsing robot urdf xml string.
[move_group-1] [INFO] [1633437090.468158900] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0019955 seconds
[move_group-1] [INFO] [1633437090.468252300] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[rviz2-3] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-ur'
[move_group-1] Link base had 0 children
[move_group-1] Link base_link_inertia had 1 children
[move_group-1] Link shoulder_link had 1 children
[move_group-1] Link upper_arm_link had 1 children
[move_group-1] Link forearm_link had 1 children
[move_group-1] Link wrist_1_link had 1 children
[move_group-1] Link wrist_2_link had 1 children
[move_group-1] Link wrist_3_link had 1 children
[move_group-1] Link flange had 1 children
[move_group-1] Link tool0 had 0 children
[move_group-1] [INFO] [1633437090.552503200] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1633437090.552735000] [moveit.ros_planning_interface.moveit_cpp]: Listening to '/joint_states' for joint states
[move_group-1] [INFO] [1633437090.554331800] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/joint_states'
[move_group-1] [INFO] [1633437090.556117000] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/move_group/planning_scene_monitor' for attached collision objects
[move_group-1] [INFO] [1633437090.556232000] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1633437090.556951200] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/move_group/publish_planning_scene'
[move_group-1] [INFO] [1633437090.556979500] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1633437090.558818600] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1633437090.560537500] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1633437090.560665700] [moveit.ros.occupancy_map_monitor]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1633437090.560710200] [moveit.ros.occupancy_map_monitor]: No 3D sensor plugin(s) defined for octomap updates
[move_group-1] [INFO] [1633437090.570554300] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'move_group'
[move_group-1] [INFO] [1633437090.612436400] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[move_group-1] [INFO] [1633437090.619465200] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1633437090.619631100] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[move_group-1] [INFO] [1633437090.619643100] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[move_group-1] [INFO] [1633437090.619697900] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[move_group-1] [INFO] [1633437090.619726900] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.050000
[move_group-1] [INFO] [1633437090.619735100] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1633437090.619753700] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[move_group-1] [INFO] [1633437090.619761300] [moveit_ros.fix_start_state_collision]: Param 'move_group.jiggle_fraction' was not set. Using default value: 0.020000
[move_group-1] [INFO] [1633437090.619788100] [moveit_ros.fix_start_state_collision]: Param 'move_group.max_sampling_attempts' was not set. Using default value: 100
[move_group-1] [INFO] [1633437090.619806900] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[move_group-1] [INFO] [1633437090.619820300] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[... 140 lines cut ...]
[move_group-1] [INFO] [1633437162.483399400] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:354 - ur_manipulator/ur_manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[move_group-1] [INFO] [1633437162.483604000] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/tools/multiplan/src/ParallelPlan.cpp:135 - ParallelPlan::solve(): Solution found by one or more threads in 0.014189 seconds
[move_group-1] [INFO] [1633437162.483902000] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:220 - ur_manipulator/ur_manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[move_group-1] [INFO] [1633437162.484012400] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:220 - ur_manipulator/ur_manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[move_group-1] [INFO] [1633437162.484104100] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:220 - ur_manipulator/ur_manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[move_group-1] [INFO] [1633437162.484131300] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:220 - ur_manipulator/ur_manipulator[RRTConnectkConfigDefault]: Starting planning with 1 states already in datastructure
[move_group-1] [INFO] [1633437162.484911200] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:354 - ur_manipulator/ur_manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[move_group-1] [INFO] [1633437162.484959800] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:354 - ur_manipulator/ur_manipulator[RRTConnectkConfigDefault]: Created 4 states (2 start + 2 goal)
[... cut ...]
[rviz2-3] [INFO] [1633437163.067073000] [move_group_interface]: time taken to generate plan: 0.0240412 seconds
[move_group-1] [INFO] [1633437212.704172100] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[move_group-1] [INFO] [1633437212.704324800] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[move_group-1] [INFO] [1633437212.705621100] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
```

Thijsons, 2021-10-05:
I had the same problem a few days ago. Managed to fix it by removing `"publish_planning_scene_topic": "/move_group/publish_planning_scene"` (line 287) from the planning_scene_monitor_parameters in the ur_moveit.launch.py file. Not quite sure why it works and if it breaks anything so I would love to know more as well.

NebuDa, 2021-10-06:
Thanks for your advise @Thijsons but I have tried it and it changes nothing. Boxes dissappear on plan after attaching and get also ignored by simple plan without attaching.
Could you verify it, please? Perhaps you changed something else, too?

NebuDa, 2021-10-06:
Just forgot it: 
I have installed the Foxy Branch on Foxy with Ubuntu 20.04 and the URCap 1.0.5

NebuDa, 2021-10-07:
@destogl I haven't done that yet because the MoveIt! Tutorial with the Panda Robot works great. I have already tried to compare the tutorial (see below) with the configuration above, but have not found any difference yet. 

`ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true`

closed 2021-10-14 by livanov93