UniversalRobots/Universal_Robots_ROS2_Driver #234: How to use dashboard?

borninfreedom, 2021-11-04:
I see there is a `ur_dashboard_client.launch.py`, I started it and there was nothing, I want to know how to use dashboard?

davidgrenner, 2021-11-16:
hey,
the dashboard_client just provides  an Interface for ROS services. 
You can  get or set information over the console or by using the 'Service Caller Plugin' in rqt

fmauch, 2021-12-09:
If you launch your robot using the usual `ros2 launch ur_bringup ur5e.launch` syntax, the dashboard server should already be running. The respective services can be seen using

```
$ ros2 service list | grep dashboard
/dashboard_client/add_to_log
/dashboard_client/brake_release
/dashboard_client/clear_operational_mode
/dashboard_client/close_popup
/dashboard_client/close_safety_popup
/dashboard_client/connect
/dashboard_client/describe_parameters
/dashboard_client/get_loaded_program
/dashboard_client/get_parameter_types
/dashboard_client/get_parameters
/dashboard_client/get_robot_mode
/dashboard_client/get_safety_mode
/dashboard_client/list_parameters
/dashboard_client/load_installation
/dashboard_client/load_program
/dashboard_client/pause
/dashboard_client/play
/dashboard_client/popup
/dashboard_client/power_off
/dashboard_client/power_on
/dashboard_client/program_running
/dashboard_client/program_saved
/dashboard_client/program_state
/dashboard_client/quit
/dashboard_client/raw_request
/dashboard_client/restart_safety
/dashboard_client/set_parameters
/dashboard_client/set_parameters_atomically
/dashboard_client/shutdown
/dashboard_client/stop
/dashboard_client/unlock_protective_stop
```

For example, to perform a break_release, you can call
```
ros2 service call /dashboard_client/brake_release std_srvs/srv/Trigger {}
```

Note, that for most of the dashboard services, the robot has to be in remote_control mode if using a real robot. Otherwise requests will be ignored / rejected. With a URSim connected instead of a real robot (or a CB3 robot) this won't be an issue.

fmauch, 2022-04-19:
Closing as this has been answered and no reaction since.

closed 2022-04-19 by fmauch