UniversalRobots/Universal_Robots_ROS2_Driver #235: Force/torque sensor broadcaster publishes wrench with incorrect orientation

schornakj, 2021-11-04:
Expected: Wrench measurement from e-Series wrist force/torque sensor is relative to whatever frame ID I set when configuring the force/torque controller plugin (for example, the `tool0` frame at the robot tool flange).

Actual: Force/torque wrench measurements are oriented relative to the robot's base link, and are consequently almost always incorrect as the robot's pose changes.

I think this is just because the UR driver's convention for reporting TCP force is to report it relative to the base coordinate system, so additional work is needed to transform it into the correct frame. This is reflected in [this guide](https://www.universal-robots.com/articles/ur/programming/weight-detection-of-any-picked-object-using-e-series/) about using a UR script program to weigh an object using the force/torque sensor.

Additional potentially-useful info:
- ROS2 Rolling
- UR ROS2 driver package commit: `349aa4f`
- UR client library package commit: `90338a7` (a few commits later than the `1.0.0` tag).
- UR Software version `5.11.0.108249`

schornakj, 2021-11-04:
Tagging @destogl since we've talked about this some already.

gavanderhoorn, 2021-11-04:
I remember discussing this with @fmauch. I'm not sure any more, but I believe he suggested transforming the measurements to the TCP, while I felt that was different from what URScript does/did, and thus the driver should not do this either.

In the end the ROS 1 version of the driver transforms measurements to the TCP frame: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/commit/00e0fbee3d845c70d77e1655f755d06004cc7cc3.

Is that no longer the case in the ROS 2 version of the code?

schornakj, 2021-11-05:
> Is that no longer the case in the ROS 2 version of the code?

Apparently the ROS 2 version of the driver lost that change! It was probably a casualty of the refactor to support ros2_control.

Now that #237 is submitted I'll add comments there instead.

PR #237 (2021-11-10): Transform force-torque into tcp frame.
Closes #235 
Sets the rotation of the wrench message into the tool center point.
closed 2021-11-10 by livanov93