UniversalRobots/Universal_Robots_ROS2_Driver #247: Exception SubstitutionFailure: exectuable ' ' not found on the libexec directory

NebuDa, 2021-11-25:
I have a new Setup:
- Ubuntu System 20.04
- UR Simulator 5.11.5
- ROS2 Galactic from Debians
- MoveIt!2 from Sources
- ROS UR Driver from Main Branch with additional Packages for MoveIt! 

After building it like the instructions here, I got the following Error on Testing, and I got this error on other test Commands also, like 
`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=127.0.0.1 launch_rviz:=false`

```
flo@NBK-18-Ubuntu:~/workspace/ros_ur_driver$ ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/flo/.ros/log/2021-11-25-09-32-13-101431-NBK-18-Ubuntu-85440
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure("executable '  ' not found on the libexec directory '/home/flo/ws_moveit2/install/controller_manager/lib/controller_manager' ")>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 2 more times]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/actions/node.py", line 484, in execute
    ret = super().execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 837, in execute
    self.__expand_substitutions(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 681, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 681, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_context.py", line 197, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/substitutions/executable_in_package.py", line 84, in perform
    raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: executable '  ' not found on the libexec directory '/home/flo/ws_moveit2/install/controller_manager/lib/controller_manager' 
[INFO] [dashboard_client-2]: process started with pid [85446]
```

Any Ideas?

OHeimann, 2021-11-25:
Hey, 
I've happened to stumble on a similiar error yesterday. I haven't found why it did occur on one machine (arm) and not on another (x86) but the manual fix for me at least was rather fast:

```
launch.substitutions.substitution_failure.SubstitutionFailure: executable '  ' not found on the libexec directory '/home/flo/ws_moveit2/install/controller_manager/lib/controller_manager' 
``` 

This line hints at an empty string for the executable passed to a node in the launch file. In my case the joint_controller_manager_spawner in ur_controller.launch.py had the executable arg set to an empty string. After looking at the other *_spawner nodes in the launch file I updated that line to also link to spawner.py. 

prior:
```python
    joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="",
        arguments=["joint_state_broadcaster", "--controller-manager", "/controller_manager"],
        condition=UnlessCondition(use_ignition),
    )
```

post fix:
```python
    joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="spawner.py",
        arguments=["joint_state_broadcaster", "--controller-manager", "/controller_manager"],
        condition=UnlessCondition(use_ignition),
    )
```

I hope this helps as a fast fix for you as well.

relffok, 2021-11-25:
Seems like this is a copy paste error of this commit https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/commit/3e099dda8c42e3f584cb10839d3a02aec2cc3f49

NebuDa, 2021-11-25:
omg have it seen by myself nine days ago and forgotten. Thanks! I leave it open until it is fixed.

closed 2021-11-25 by livanov93