UniversalRobots/Universal_Robots_ROS2_Driver #252: Invalid path in git

michaelaeriksen, 2021-12-02:
When checking out this repo I get 

```
error: invalid path 'ur_robot_driver/resources/ursim_driver/ursim/aux/.vol/urcap_ros_control.installation'
```

urrsk, 2021-12-03:
@michaelaeriksen Thanks for your question.
I am not really sure why windows is not happy with that filename, but thanks for the insight
Maybe you can get going with Windows Subsystem for Linux (WSL). 

Please note that the driver does not yet support Windows natively.

michaelaeriksen, 2021-12-03:
I forked the `Universal_Robots_ROS2_Driver` repo in wsl and removed the files giving me problems in Windows. I have the ROS version of the drivers working on Windows natively and I'm now looking at migrating to ROS2. A while back I forked the `Universal_Robots_Client_Library` repo and changed all the socket code to use the platform independent library asio and the ROS version of the drivers are working great.

urrsk, 2021-12-07:
@michaelaeriksen that sounds great. Have you any idea of why windows did not like that file?
We have talked about making the ROS2 driver windows compatible, and here you work sounds very interesting.
Is it possible for you to share the ```asio``` version?

relffok, 2022-04-21:
`aux` seems to be a reserved filename on windows. We ran into this issue before and renaming the path worked for us! 

https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file?redirectedfrom=MSDN

fmauch, 2022-04-21:
> `aux` seems to be a reserved filename on windows. We ran into this issue before and renaming the path worked for us!
> 
> https://docs.microsoft.com/en-us/windows/win32/fileio/naming-a-file?redirectedfrom=MSDN

Thanks for that insight!

haellingsen, 2022-09-21:
How did you workaround this? Did you fork and rename?

fmauch, 2022-09-21:
@haellingsen you could use the branch from #478 for example. This will see some more work today, but the problematic folder is already removed.

appselskapet, 2022-09-21:
> @haellingsen you could use the branch from #478 for example. This will see some more work today, but the problematic folder is already removed.

Thanks.

Although I ran git config core.protectNTFS false and cloned it.

fmauch, 2023-05-09:
Since the path is no longer part of the main branch, I'm going to close this.

closed 2023-05-09 by fmauch