UniversalRobots/Universal_Robots_ROS2_Driver #253: Remote control of a Robotiq Gripper connected to the UR5 via USB

Tonidor, 2021-12-08:
# Problem
Currently, we are trying to control the Robotiq Gripper which is connected via USB to a UR5 remotely using the Universal_Robots_ROS2_Driver but couldn't manage to do so. Regarding this topic, we would appreciate some instructions.

# Setup
PC OS: Ubuntu 20.4
ROS Distribution: Foxy
UR model: UR5

# What we tried

Apart from using the ROS2 driver, we tried using the [URX library](https://github.com/SintefManufacturing/python-urx). The approach of this library is to connect via a TCP socket on port 30002 and send a URScript that controls the gripper. This was successful. But if we run the ROS2 driver and send a trajectory to the UR5n and start controlling the gripper using the URX library, the driver loses connection. This gets printed in the consol where the driver runs:

> [ros2_control_node-1] [INFO] [1638961383.669717614] [UR_Client_Library]: Connection to reverse interface dropped.

In order to make sure, there is nothing else happening in the URX library that distracts the ROS2 driver, we extracted a minimal example in python on how the gripper is controlled:

### gripper_control.py

> import socket
> from time import sleep
> 
> HOST = "192.168.1.102"
> SEC_PORT = 50002
> 
> i_sec = socket.create_connection((HOST, SEC_PORT), timeout=0.5)
> 
> with open('gripper_close.txt') as f:
>     close_program = f.read()
> 
> i_sec.send(str.encode(close_program))

And this is the URScript that is loaded from 'gripper_close.txt' (the code between myProg and end should be indented):

### gripper_close.txt

> def myProg():
> 	socket_close("gripper_socket")
> 	socket_open("127.0.0.1",63352,"gripper_socket")
> 	set_analog_inputrange(0,0)
> 	set_analog_inputrange(1,0)
> 	set_analog_inputrange(2,0)
> 	set_analog_inputrange(3,0)
> 	set_analog_outputdomain(0,0)
> 	set_analog_outputdomain(1,0)
> 	set_tool_voltage(0)
> 	set_runstate_outputs([])
> 	set_payload(0.85)
> 	socket_set_var("SPE",255,"gripper_socket")
> 	sync()
> 	socket_set_var("FOR",50,"gripper_socket")
> 	sync()
> 	socket_set_var("ACT",1,"gripper_socket")
> 	sync()
> 	socket_set_var("GTO",1,"gripper_socket")
> 	sync()
> 	sleep(0.1)
> 	socket_set_var("POS",255,"gripper_socket")
> 	sync()
> 	sleep(2.0)
> end

Still with this minimal version, the ROS2 driver loses connection, when we run this example when sending a trajectory through the ros interface.

We haven't tried that yet but we are also thinking about forwarding the port of the grippe (63352), so we can access it remotely.

# What we understand so far

There are different ports used for communicating with the UR:

**50002**: primary interface, remote computer sends URScript to the UR that is interpreted by the URCap, 125Hz
**50001**: UR requests the the remote computer to send URScript
**30003**: UR sends the current status of the robot in real-time
**30002**: secondary interface, remote computer sends URScript that contains commands that are not interfering with commands from the primary interface, 10Hz
**54321**: port of the tool communication interface for a remote computer to use tools connected to the UR?, not available for the UR5 but for UR5e?

Is this interpretation correct? Is there anything else happening on these ports that is important for our problem?

We searched in the ROS2 driver repository for more clues on why using port 30002 from another program interferes with the ROS2 driver but we couldn't find a clue. Maybe we missed something.

Ideally, we would like to use a controller with a topic where we can publish control messages for the Robotiq gripper.
Is this possible or are there any other workarounds to achieve remote control of the gripper?

We would much appreciate some tips and explanations.

benediktkreis, 2022-02-21:
Are there any updates? I'm facing a similar issue with the OnRobot RG2-FT gripper. You can find my post in [ROS Answers](https://answers.ros.org/question/396393/control-and-read-sensors-of-onrobot-rg2-ft-gripper-mounted-to-a-ur5-robot-arm-using-ros2/). I'm grateful for any help.

Tonidor, 2022-02-21:
After a long time trying different approaches, we managed to find a solution that works.
We switched to using the ur_rtde library which enabled simultanious remote control of the gripper and the UR5 robot.

Here are some related links:

- [installation guide](https://sdurobotics.gitlab.io/ur_rtde/installation/installation.html#quick-install)
- [robotiq gripper guide](https://sdurobotics.gitlab.io/ur_rtde/guides/guides.html#use-with-robotiq-gripper)
- [gitlab repo](https://gitlab.com/sdurobotics/ur_rtde)

For us, Option 3 of connecting to the gripper mentioned in the robotiq gripper guide worked. Although what we do different than in the tutorial is to activate the gripper on the UR5 control pad manually and leave out the code line `gripper.activate()`. Somehow the script seems to get stuck on this point so we switched to our workaround.

If this alternative is suitable and works for you too, I would be curious to know. :)
Feel free to ask if you have any further questions regarding the topic!

benediktkreis, 2022-02-21:
Thanks for your quick reply. So you aren't using the Universal_Robots_ROS2_Driver at all?

Since I would like to use Moveit2 and other ROS2 packages, I could use the Universal_Robots_ROS2_Driver to control the arm and the ur_rtde library to control the gripper and wrap the required code into a ROS node, right?

Tonidor, 2022-02-21:
Yes, we don't use the driver anymore. I'm not sure weather it's possible to use the driver and the ur_rtde library together since I've experienced that the driver stops working when I tried other approaches (like using sockets) to control the gripper simultaneously. I would expect that it doesn't work because the driver kind of "compromises" many (all?) ports on the robot. Although, it would be interesting to know if it works in case of the ur_rtde library. I have no experience with Moveit2 but I would probably try to wrap the ur_rtde library with ROS2 and use it together with Moveit2 rather than mixing control methods (driver + ur_rtde).

fmauch, 2022-02-23:
Just as a sidenote: This driver implements and uses the RTDE interface.

`ur_rtde` seems to have a wrapper for robitiq grippers. As you can see in their [gripper example](https://gitlab.com/sdurobotics/ur_rtde/blob/master/examples/cpp/robotiq_gripper_example.cpp) it connects to the modbus port directly. So, you should be able to use this driver and the gripper connection side-by-side.

Tonidor, 2022-02-23:
That's good to know, thanks for the comment!

Abelpy1, 2022-08-30:
@anuj018 ,  @fmauch , 
Hey, recently i tried to control rg2 gripper with ur_rtde library following https://gitlab.com/sdurobotics/ur_rtde/-/issues/108 and https://sdurobotics.gitlab.io/ur_rtde/examples/examples.html#use-with-custom-script .
In this way, i can't use Universal_Robots_ROS2_Driver any more once the ur_rtde is running.
I am curious that if you have solved the problem or get more informations?
Hope for your reply.

urrsk, 2022-09-02:
@Abelpy1 No we have not solved the ur-rtde(made by SDU)/ROS conflict.
I am not 100% sure, but it is most likely because we both try to control the digital outputs. This is the same reason why the ROS driver and fieldbus's (ProfiNet and EtherNet support) does not work together.

But after I have read this thread again, I thing that there are different other possible workarounds:
1. Use the External Control URCap for injecting the ROS driver scriptcode together with other logic in a  Polyscope program e.g. listen at the digital outputs and if digital_input[0] is high the close the gripper use the URCap node that Robotiq have made
(Polyscope is the teachpendant's UI).
2. Another could be to launch the ROS driver with a customise the scriptcode, see driver [doc](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/doc/ROS_INTERFACE.md#script_filename-required) . In addition, please note as well that the ROS and ROS2 driver is build on top of the Universal_Robots_Client_Library and therefore the urscript that the driver uses, is placed in the repo: https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/resources/external_control.urscript
3. Forward the tty from the robot to the robotiq's ROS driver using a usb dongle (maybe you need a ROS to ROS2 bridge as well)
The UR's [ToolComm_Forwarder_URCap](https://github.com/UniversalRobots/Universal_Robots_ToolComm_Forwarder_URCap/blob/master/src/main/resources/com/fzi/rs485/impl/daemon/daemon-rs485.py) is hard coded for the tool tty. A workaround could be to change that parameter to your custom need.
This could be a feature for the Forward urcap, so please let me know if any are succeeding using this approach.