UniversalRobots/Universal_Robots_ROS2_Driver #254: Moveit with ursim

Abelpy1, 2021-12-09:
Hello, recently I tried to test `Moveit` on  `ursim`, and my solutions are as follows:
### My Environment:
- docker
- ROS Distro: foxy

### Steps to test:
1. Firstly, referring from `Universal_Robots_ROS2_Driver/ur_robot_driver/resources/ursim_driver` , I created two containers, one for ursim, another for experiments.
2. Then i run `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=172.17.0.2 use_fake_hardware:=false launch_rviz:=false` to start the robot driver and controllers, connect to the ursim. 
3. After that, using `Moveit2` package, I wrote code to implement `pose tracking`, expect to control the movement of the robot arm with the handle. But it failed. When i moved the handle, the robot didn't move at all. 

What do you think may have caused the experiment to fail? I check the output of some topics, like `/joint_trajectory_controller/joint_trajectory`, it looks normal.  I am very confused that if the ursim doesn't support `URPositionHardwareInterface`？

destogl, 2021-12-09:
@Abelpy1 this doesn't seem to be a problem with this repository. Am I right?

Can you control the robot with MoveIt only?

fmauch, 2021-12-09:
I've just experiencing the same thing when using the [URSim docker image](https://hub.docker.com/r/universalrobots/ursim_e-series).

The robot's logs show that it tries to connect to the reverse interface using the IP `127.0.0.1` due to the way I setup using the docker container.

The current parameter structure doesn't support setting the IP address that should be used for accessing the ROS machine. In the client library we have the parameter [`reverse_ip`](https://github.com/UniversalRobots/Universal_Robots_Client_Library/blob/master/src/ur/ur_driver.cpp#L57). But that's currently not available in the ROS2 driver, yet.

I'm still rather new to the whole docker network setup topic, but I'll start using it more frequently in the near future.

fmauch, 2023-05-09:
Since a script for starting URSim is now included into the package and usage is [documented](https://docs.ros.org/en/rolling/p/ur_robot_driver/usage.html#usage-with-official-ur-simulator), I'm going to close this.

closed 2023-05-09 by fmauch