UniversalRobots/Universal_Robots_ROS2_Driver #256: Test tool communication

livanov93, 2021-12-09:
@fmauch After discussion could you please give us some feedback on testing tool communication after #218 has been merged.
Thanks!

fmauch, 2021-12-09:
yep, I'll do that.

fmauch, 2021-12-10:
I wasn't able to test the full pipeline on a real robot as I couldn't find the connection cable for my hardware in order to connect it to the UR. The script itself should work though, as it is the same that is used for the ROS1 driver.

However, I noticed that the tool_communication script does not get started automatically, when `use_tool_communication` is set to `true` in the launchfile. For the ROS1 driver we actually [started the script](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/master/ur_robot_driver/launch/ur_control.launch#L61) when that argument was set.

livanov93, 2021-12-15:
Sounds good to start the script if the parameter is set to `true`. I will address this in a follow-up PR

closed 2022-02-23 by destogl