UniversalRobots/Universal_Robots_ROS2_Driver #262: Having trouble running the driver on Rolling

fmauch, 2021-12-10:
When trying to run the driver on ROS2 Rolling I experienced some problems. (See launchfile output below)

My workspace looks like this and is up-to-date (as of writing this):
```
========= Universal_Robots_Client_Library =========
## master...origin/master
========= Universal_Robots_ROS2_Driver =========
## main...origin/main
========= ur_msgs =========
## ros2...origin/ros2
```

When using the driver with ROS2 Foxy (with the `foxy` branch of this driver), everything works out great. I am currently dropping this here, without further investigating it. If it works for everybody else, feel free to close this, but since I just installed ROS Rolling and tested the driver, I guess others might also fall into this problem.

```
$ ros2 launch ur_bringup ur5e.launch.py robot_ip:=192.168.56.101 launch_rviz:=false
[INFO] [launch]: All log files can be found below /home/mauch/.ros/log/2021-12-10-11-43-24-296701-ids-felino-155786
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [155792]
[INFO] [ros2_control_node-1]: process started with pid [155790]
[INFO] [robot_state_publisher-3]: process started with pid [155794]
[INFO] [spawner-4]: process started with pid [155796]
[INFO] [spawner-5]: process started with pid [155798]
[INFO] [spawner-6]: process started with pid [155800]
[INFO] [spawner-7]: process started with pid [155802]
[INFO] [spawner.py-8]: process started with pid [155804]
[INFO] [spawner-9]: process started with pid [155806]
[dashboard_client-2] [INFO] [1639133005.376358381] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2]
[robot_state_publisher-3] Link base had 0 children
[robot_state_publisher-3] Link base_link_inertia had 1 children
[robot_state_publisher-3] Link shoulder_link had 1 children
[robot_state_publisher-3] Link upper_arm_link had 1 children
[robot_state_publisher-3] Link forearm_link had 1 children
[robot_state_publisher-3] Link wrist_1_link had 1 children
[robot_state_publisher-3] Link wrist_2_link had 1 children
[robot_state_publisher-3] Link wrist_3_link had 1 children
[robot_state_publisher-3] Link flange had 1 children
[robot_state_publisher-3] Link tool0 had 0 children
[robot_state_publisher-3] [INFO] [1639133005.382554600] [robot_state_publisher]: got segment base
[robot_state_publisher-3] [INFO] [1639133005.382671008] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1639133005.382688595] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-3] [INFO] [1639133005.382698174] [robot_state_publisher]: got segment flange
[robot_state_publisher-3] [INFO] [1639133005.382707446] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1639133005.382716342] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1639133005.382724953] [robot_state_publisher]: got segment tool0
[robot_state_publisher-3] [INFO] [1639133005.382733630] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-3] [INFO] [1639133005.382742373] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-3] [INFO] [1639133005.382751494] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-3] [INFO] [1639133005.382760230] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1639133005.389215072] [URPositionHardwareInterface]: Starting ...please wait...
[ros2_control_node-1] [INFO] [1639133005.389285621] [URPositionHardwareInterface]: Initializing driver...
[ros2_control_node-1] [WARN] [1639133005.389975022] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[ros2_control_node-1] [INFO] [1639133005.593252395] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ros2_control_node-1] [INFO] [1639133005.593525285] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[ros2_control_node-1] [WARN] [1639133006.610225421] [UR_Client_Library]: DEPRECATION NOTICE: Passing the calibration_checksum to the UrDriver's constructor has been deprecated. Instead, use the checkCalibration(calibration_checksum) function separately. This notice is for application developers using this library. If you are only using an application using this library, you can ignore this message.
[ros2_control_node-1] [WARN] [1639133006.611226610] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[ros2_control_node-1] [ERROR] [1639133007.645380835] [UR_Client_Library]: The calibration parameters of the connected robot don't match the ones from the given kinematics config file. Please be aware that this can lead to critical inaccuracies of tcp positions. Use the ur_calibration tool to extract the correct calibration from the robot and pass that into the description. See [https://github.com/UniversalRobots/Universal_Robots_ROS_Driver#extract-calibration-information] for details.
[ros2_control_node-1] [WARN] [1639133007.645513607] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[ros2_control_node-1] [INFO] [1639133007.645786238] [URPositionHardwareInterface]: System successfully started!
[ros2_control_node-1] [INFO] [1639133007.653296234] [controller_manager]: update rate is 530 Hz
[ros2_control_node-1] [ERROR] [1639133007.655390414] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.657275263] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.659270052] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.661199951] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.663148316] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.665072120] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.667011140] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.668952037] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.670895134] [URPositionHardwareInterface]: Unable to write to hardware...
[... 114 lines cut ...]
[ros2_control_node-1] [ERROR] [1639133007.852330272] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.854345566] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.856189997] [URPositionHardwareInterface]: Unable to write to hardware...
[spawner-9] [INFO] [1639133007.857395457] [spawner_joint_trajectory_controller]: Configured and started joint_trajectory_controller
[ros2_control_node-1] [ERROR] [1639133007.858248055] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.860163168] [URPositionHardwareInterface]: Unable to write to hardware...
[spawner-7] [INFO] [1639133007.860941589] [spawner_force_torque_sensor_broadcaster]: Configured and started force_torque_sensor_broadcaster
[ros2_control_node-1] [ERROR] [1639133007.863397339] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.865440986] [URPositionHardwareInterface]: Unable to write to hardware...
[spawner-6] [INFO] [1639133007.866359426] [spawner_speed_scaling_state_broadcaster]: Configured and started speed_scaling_state_broadcaster
[ros2_control_node-1] [ERROR] [1639133007.867335261] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.869294381] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.871248701] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.873167235] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.879381378] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.881246414] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.883313093] [URPositionHardwareInterface]: Unable to write to hardware...
[... cut ...]
[ros2_control_node-1] [ERROR] [1639133007.887333144] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.889384208] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.891459385] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.893428785] [URPositionHardwareInterface]: Unable to write to hardware...
[ros2_control_node-1] [ERROR] [1639133007.895511664] [URPositionHardwareInterface]: Unable to write to hardware...
[... pressed CTRL-C at some point]
```

christianlandgraf, 2021-12-10:
I'm getting the same error but it disappears if the External Control URCap is running. If it is stopped, the error messages reappear

fmauch, 2021-12-16:
I can confirm that once I start running the external control program the message goes away. Looking at the code I am wondering why this message was even added.

When designing the interface we deliberately chose only to send commands to the hardware if it could receive them (meaning if the custom port for receiving commands is actually open). In the ROS1 driver we [silently ignore that case](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/blob/c3378599d5fa73a261328b326392e847f312ab6b/ur_robot_driver/src/hardware_interface.cpp#L734) not out of laziness but as this would actually be the expected state.

That behaviour had been added in 349aa4f711dc416304d1a93a0f89192e9bf1ff8a. While it makes sense on the reading side, it does not make too much sense on the writing side from my point of view. @livanov93 am I missing something?

destogl, 2021-12-22:
Other than that, it appears that you are missing `forward_position_controller`

fmauch, 2022-06-27:
I think, this can be closed by now...

closed 2022-06-27 by fmauch