UniversalRobots/Universal_Robots_ROS2_Driver #265: a lot Parameters have invalid type: is of type {double}, setting it to {string} is not allowed

NebuDa, 2021-12-16:
I have built ROS2 Galactic, Moveit2 and Main Branch of this driver on Ubuntu. By starting like the tutroial in the readme with MoveIt I got the following error on different parameters:
```
[rviz2-3] [INFO] [1639652465.916282193] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00852234 seconds
[rviz2-3] [INFO] [1639652465.917162913] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[rviz2-3] [ERROR] [1639652465.936346953] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.shoulder_pan_joint.max_position' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.shoulder_pan_joint.max_position} is of type {double}, setting it to {string} is not allowed.
```
Another parameter is by example: robot_description_kinematics.ur_manipulator.kinematics_solver_timeout

Is there anything wrong with Ubuntu or ROS2 configuration, that it tries to load it as string?

destogl, 2021-12-22:
I am guessing that you are using rolling. In any case, please check #268 if it solves your problem and comment there.

fmauch, 2022-06-20:
This should be resolved by now.

closed 2022-06-20 by fmauch